amymcgovern/pyparrot #5: fly_direct doesn't work over wifi

antonkho, 2017-12-06:
I downloaded and ran Mambo demos using my Windows computer connected to Mambo drone via wifi.
Both in demoMamboTricks and in demoMamboDirectFlight mambo.fly_direct doesn't seem to have any effect on the drone. mambo.flip seem to work fine.
I'm running pyparrot-1.1 as downloaded from "release" tab.

amymcgovern, 2017-12-06:
Are you also running the latest firmware on your mambo?

anton-kh, 2017-12-07:
What is the best way to check that and update if needed? I did update the
firmware via the iPhone app, but I don't know if that made it the latest.

amymcgovern, 2017-12-07:
It should have.  I'm pretty stumped on the issues.  I have run it on 9 separate mambos and they all perform fine.  I have sometimes seen the flip commands ignored but is dependent on the battery status.  I have never seen it ignore the fly_direct commands.  Does it print any errors?

anton-kh, 2017-12-08:
I don't see any errors, also take off/landing/flips work just fine. I'm also running 64 bit python, if that makes any difference. Could you point me to some documentation on what is called "PCMD"? I could maybe check if packets sent to the drone have the right formatting via a couple of additional print statements.

amymcgovern, 2017-12-08:
Unless you have changed the code, there is no reason for the packets to be broken.  The same code is working fine on my Mambos... PCMD is documented in minidrone.xml.

closed 2018-02-08 by amymcgovern