amymcgovern/pyparrot #11: Could not find some sensor : Bebop2

tomoyuki1111, 2018-01-21:
I am looking for python code for bebop2, and I find your source, thank you. 
I tried "demoBebopIndoors.py", but video streaming doesn't work and these comment appears:
  _"Could not find sensor in list -ignoring for now. Packet infobelow "_
  _(142,  0, 3)_
The header_tuple=(134, 0, 3), (134, 0, 4),(144, 0, 2), (144, 0,1) are also appeared, 
and one Error comment "_Error:tried to parse a bad sensor packet_"
I don't know the cause, but sensor_name is none maybe.
However, other commands such as takeoff and land works.

amymcgovern, 2018-01-21:
This is not a bug.  This is reporting a sensor that I am not currently parsing.  I know about it.  it isn't an important sensor which is why I put it as a lower priority item to get to.

closed 2018-01-21 by amymcgovern

tomoyuki1111, 2018-01-22:
Thank you for your kind replay. I understand it.
To erase these comments appropriately, I will try to add some code for the other xml file data to project_xml.