amymcgovern/pyparrot #13: send_auto_takeoff_command() missing in wifiConnection.py

srijith-vijay, 2018-02-07:
Hi,

I was going through the wifiConnection.py file and could not find the function 'send_auto_takeoff_command()' function which is internally called by the 'turn_on_auto_takeoff()' function. However, this function is present in the bleConnection.py file. 

Was this done on purpose to support auto-takeoff feature with only BLE? I updated the wifiConnection.py file, at my end, to include this function and the throw mode seems to work fine.  

However, the next time I connect the drone and create the Mambo instance, I observe that the drone stops sending the initial packets after sending the "cut-out enabled" packet. 
I am able to take off and perform other operations however the drone does not send any sensor data.

amymcgovern, 2018-02-07:
I tried to ensure that each function exists in BLE and wifi but I guess I missed that one.  I'll update it shortly.  Also, I don't know why you would be dropping sensor data.  I am focusing mostly on the wifi since it sends a lot more sensor information but I've not seen it fail to send any unless you send an invalid packet (then it gets unhappy inside the drone).  I'll update the function and test shortly.

srijith-vijay, 2018-02-07:
I double checked that the packets being sent are valid. If it was invalid, the functions like takeoff, flip, land etc. would not be working. I am able to execute all these operations. However, the drone remains silent in terms of sending the sensor data every 0.5 seconds.

Interestingly, I observe this only on my laptop and not on my desktop computer. I am wondering if this has something to do with the controller name/type being sent in the initial JSON packet.

Let me know if you face the same issue after executing the throw mode and creating the mambo instance again.

amymcgovern, 2018-02-07:
That is very odd.  I'll hunt it down tonight.  I don't have a drone with me at work today!

amymcgovern, 2018-02-08:
I just put the command into the wifi and it seems to work just fine.  Sends lots of sensor info to me.  Can you update and try again?

srijith-vijay, 2018-02-08:
I tried to reproduce it again and this time it worked fine. The issue seems to be sporadic.
Will update you with the reproduction steps, in case I run into the issue again. Thanks!

amymcgovern, 2018-02-08:
Glad to hear it!

closed 2018-02-08 by amymcgovern