amymcgovern/pyparrot #15: Problem with Vision

erhart, 2018-02-16:
I can't start the DroneVision.open_video. 
ffmpeg can't connect to my Bebop 2. 

I'm using Archlinux with python 3.6, is it maybe the firmware version of my drone?
The drone is new, so the fw version is 4.4.

amymcgovern, 2018-02-16:
Send an error message please.  Mine is also 4.4

erhart, 2018-02-19:
Service Bebop2-117628._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-117628._arsdk-090c._udp.local.', address=b'\xc0\xa8*\x01', port=44444, weight=0, priority=0, server='Bebop2-117628.local.', properties={b'{"device_id":"PI040384AH7H117628"}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at 
 Success in setting up the wifi network to the drone! 
59
/home/Code/pyparrot/images
/home/Code/pyparrot/utils
Opening ffmpeg
ffmpeg -protocol_whitelist "file,rtp,udp" -i /home/Code/pyparrot/utils/bebop.sdp -r 30 image_%03d.png &
Opening non-blocking readers
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers

  built with gcc 7.3.0 (GCC)

  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-avisynth --enable-avresample --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libass --enable-libbluray --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-shared --enable-version3 --enable-omx

  libavutil      55. 78.100 / 55. 78.100

  libavcodec     57.107.100 / 57.107.100

  libavformat    57. 83.100 / 57. 83.100

  libavdevice    57. 10.100 / 57. 10.100

  libavfilter     6.107.100 /  6.107.100

  libavresample   3.  7.  0 /  3.  7.  0

  libswscale      4.  8.100 /  4.  8.100

  libswresample   2.  9.100 /  2.  9.100

  libpostproc    54.  7.100 / 54.  7.100

Invalid return value 0 for stream protocol

timeout - trying again
timeout - trying again
    Last message repeated 1 times

[sdp @ 0x55cf55513a60] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size

Consider increasing the value for the 'analyzeduration' and 'probesize' options

Input #0, sdp, from '/home/Code/pyparrot/utils/bebop.sdp':

  Duration: N/A, bitrate: N/A

    Stream #0:0: Video: h264, none, 90k tbr, 90k tbn, 180k tbc

Output #0, image2, to 'image_%03d.png':

Output file #0 does not contain any stream

Having trouble connecting to the camera 1.  A reboot of the mambo may help.
timeout - trying again
timeout - trying again
timeout - trying again
timeout - trying again
timeout - trying again
timeout - trying again
 disconnecting

erhart, 2018-02-19:
Tried to run the samples form Parrot. 
Bow I get new errors:

Service Bebop2-117628._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-117628._arsdk-090c._udp.local.', address=b'\xc0\xa8*\x01', port=44444, weight=0, priority=0, server='Bebop2-117628.local.', properties={b'{"device_id":"PI040384AH7H117628"}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at 
 Success in setting up the wifi network to the drone! 
59
/home/erhart/Arbeit/Veranstaltungen/SWTP_SS18/Code/pyparrot/images
/home/erhart/Arbeit/Veranstaltungen/SWTP_SS18/Code/pyparrot/utils
Opening ffmpeg
ffmpeg -protocol_whitelist "file,rtp,udp" -i /home/erhart/Arbeit/Veranstaltungen/SWTP_SS18/Code/pyparrot/utils/bebop.sdp -r 30 image_%03d.png &
Opening non-blocking readers
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers

  built with gcc 7.3.0 (GCC)

  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-avisynth --enable-avresample --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libass --enable-libbluray --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-shared --enable-version3 --enable-omx

  libavutil      55. 78.100 / 55. 78.100

  libavcodec     57.107.100 / 57.107.100

  libavformat    57. 83.100 / 57. 83.100

  libavdevice    57. 10.100 / 57. 10.100

  libavfilter     6.107.100 /  6.107.100

  libavresample   3.  7.  0 /  3.  7.  0

  libswscale      4.  8.100 /  4.  8.100

  libswresample   2.  9.100 /  2.  9.100

  libpostproc    54.  7.100 / 54.  7.100

Invalid return value 0 for stream protocol

    Last message repeated 1 times

[h264 @ 0x55725681e760] SPS unavailable in decode_picture_timing

[h264 @ 0x55725681e760] non-existing PPS 0 referenced

[h264 @ 0x55725681e760] SPS unavailable in decode_picture_timing

[h264 @ 0x55725681e760] non-existing PPS 0 referenced

[h264 @ 0x55725681e760] decode_slice_header error

[h264 @ 0x55725681e760] non-existing PPS 0 referenced
...

amymcgovern, 2018-02-19:
For the first one, where it says reboot (clearly I need to remove the mambo from that print statement): I have seen repeated issues where rebooting the camera is the only solution.  For the second one, I have not seen the invalid return value error!  It looks to me like it actually started though.  All of the h264 stuff is from ffmpeg and it usually prints a fair amount of that out while running!  Is it writing pictures to your images directory?

erhart, 2018-02-19:
Hello @amymcgovern ,
i don't know what i did to get form my first post to the second, but now it works. 
I did not know, that the h264 stuff was from ffmpeg. After 10 to 15 seconds the stream starts and works fine.

Thank you for this nice project!

closed 2018-02-19 by erhart