amymcgovern/pyparrot #16: Yaw value question

freaad, 2018-02-22:
Hello  Captain.
Do you have any idea that we can get the interneal IMU value such as gyro and  compass. I try to do the velocity control but sometime yaw is messing during the fluctation.
Thank you for your nice library.

amymcgovern, 2018-02-24:
I agree.  My suggestion is to go on the parrot development forum and ask them for the data.  I also would like it!  But they need to hear from the users to know that we need it.

freaad, 2018-02-28:
I asks one time but I actually do not get any answer from them. Anyway Thank you for answer.
I  am actually use the https://marvelmind.com/ this beacon system with your bebop. I used to use the pixhawk drone but bebop is more small and easy to fly in the indoor scenario.  So I am using this. When I know how to use the github well I will put on the my entire code to make branch. I honestly do not know how to manage my repository.

amymcgovern, 2018-02-28:
that is really cool!  They mostly ignore when I ask for more sensor data but I figure that the more of us who ask, the more they know we want it.

freaad, 2018-03-19:
Do you think that it is related my question?
http://developer.parrot.com/docs/bebop/#a-name-common-commonstate-sensorsstateslistchanged-a

amymcgovern, 2018-03-19:
Maybe?  Is it actually sending that back?  If it is, it is in the dictionary.  I'm saving all things in the common states. I'm just skipping a few and they are not in common (which that is)

freaad, 2018-03-19:
Ok I will look into it.

closed 2018-03-26 by amymcgovern