amymcgovern/pyparrot #17: connection failed: bebop 1

Victor804, 2018-03-02:
Hello,
I can't connect to my bebop 1.
`connecting
 connection failed: did you remember to connect your machine to the Drone's wifi network? 
False
`
I am well connected to my drone: p
And I use a raspberry.
I do not know what's the problem?
If you have an idea...

Thank you

amymcgovern, 2018-03-02:
I do not have access to a bebop 1. The documentation all days bebop 2.  Does the 1 run the same firmware?  If so, I imagine it should work.  So please ensure you are running the latest firmware.

Victor804, 2018-03-02:
I have the latest firmware for bebop 1 (4.0.6).
But the latest firmware for the bebop 2 is 4.4.2.

Thank you for answered.

lukask86, 2018-03-06:
Hi,
I am receiving the same error:
connection failed: did you remember to connect your machine to the Drone's wifi network?
I have the bebop 2 with 4.4.2, Python 3, Windows 10 and the correct wifi network.
Do you have any idea?

amymcgovern, 2018-03-06:
I am assuming both of you have connected to the wifi for the drone and waited a few seconds?  For me, I sometimes see it taking 10-30 seconds before it finally connects.

lukask86, 2018-03-06:
I also tried 30-60 seconds and nothing happened

amymcgovern, 2018-03-06:
It just keeps giving you the same error even after 60 seconds of being connected to the wifi?

lukask86, 2018-03-06:
I connected the WiFi, waited a couple of minutes and tried the connect command in python (with 30 seconds) ... still receive the error

amymcgovern, 2018-03-06:
That is very strange.  I have never observed it refusing over the long term.  I have sometimes had to reboot the bebop but then it works when it comes back (since we can't login to their computer inside the bebop I assume it just somehow came up into a funny state).

lukask86, 2018-03-06:
I tried it with 2 different PCs (both windows 10 and python 3).

amymcgovern, 2018-03-09:
Ok that is a good update.  I wonder why?  Let me see if I can install a virtual windows 10 environment.  I do all my development and testing on my Mac.

lukask86, 2018-03-09:
I installed an Ubuntu VM on my Windows PC, but it did not work...

freaad, 2018-03-11:
In the windows, the major problem in your device driver. Every times your windows major update, you should check about your lan card driver. My lancard was intel series and it has error about wifi device connection problem. After update your device driver properly it works ok. I hope this tip will help you.

lukask86, 2018-03-14:
Hi, I updated my wlan driver, but it still does not work (windows machine). 
I organized an Ubuntu 16 Notebook. I can connect to drone but after the connection I receive
"timeout - trying again"
I restarted the drone several times, but it is still not working.

freaad, 2018-03-14:
Hello. Are you using the bebop2 or not. I run the 2 types of notebook one is ubuntu the other is windows. WIndows you need to check the firewall and also device driver recent version. Second Ubuntu you need to install device driver. I also want to know that your bebop can connect with smartphone?

lukask86, 2018-03-14:
Hi, I am using the bebop 2. I disabled the firewall at my ubuntu pc and it is now working, i.e. I can control the drone. Nevertheless, when I tried to ask_for_state_update() it says "could not find sensor in list" several times. Also the Vision demo does not work...

amymcgovern, 2018-03-14:
The sensor is not an error.  I'm skipping the parsing of some of them.  I forget which ones they are right now but they are not important ones.  Nonetheless, I will address that at some point soon.

As for vision not working, what does it mean to not work?  Still saying it isn't connecting?  Are you running the bebop demo?  It doesn't fly during the demo.  It says that specifically.

freaad, 2018-03-14:
Video demo in my experience you need to wait certain time. I have to wait 10 to 30 sec to start for the image saving.

amymcgovern, 2018-03-14:
freaad is that the case with the latest code?  I tried to speed it up a lot in the last fix.  I don't think I made it a major release.  Just a bug fix.  I've been brainstorming ways to get it straight into opencv again without the bugs that opencv introduced and without the delay that the current approach introduces.  Haven't solved it yet or I would have released it.

freaad, 2018-03-14:
Windows and linux has different speed. I have no idea about why but both has delay to get the images.

amymcgovern, 2018-03-14:
Odd.  I switched the threading recently to immediately start the ffmpeg and then the vision thread that reads the ffmpeg output and *then* start the parser to see if ffmpeg started right.  The main delay is that ffmpeg takes 10-30 seconds and then the vision thread was behind.  This way, so long as it is succeeds in starting, it ends up not being behind.  But I agree it can take awhile.  It partly has to do with the I-frames.  It sends I-frames only every 250 frames, which means that it can be up to 10 seconds while it waits for a new I-frame.

freaad, 2018-03-14:
First start requires times after that it is fine but I am not sure that it catch up the realtime because if it is not I can not flying with real time  object avoidance system

amymcgovern, 2018-03-14:
It is catching up quickly but there is an inherent delay in writing out to disk and then reading it in.  Disk speeds are not 0.

freaad, 2018-03-14:
I understand but if we get the image as object we can put the image in the real-time system. I consider to use your image callback function. but I don't test it. I was need to send my bebop to exchange the new one.

amymcgovern, 2018-03-14:
The callback will let you do that.  The main issue is that opencv has a bug reading the stream from RTP and RTSP.  ffmpeg seems to be the only viable workaround.

amymcgovern, 2018-04-20:
That's good to know!!

amymcgovern, 2018-04-20:
Does it work well to fly it?  I don't have access to a bebop 1 to test.

Victor804, 2018-04-20:
I do not try to fly it but it connects well. I'll keep you informed ;)

lukask86, 2018-05-23:
Hi, I have the following situation:
1) I connect to the drone
2) I do something with the drone
3) connection gets lost due to wifi connection problems
Is there a way to check if there is a connection to the drone and if not to reconnect, i.e. how can I ask the drone about the connection state?

amymcgovern, 2018-05-23:
Since this thread has lots of people on it, @lukask86 are you flying a bebop or a mambo?  You might look at the safe_send which does a try/catch on the sending actual data to the drone and recreates the socket if it dies.  Or do you need to do a new handshake?

lukask86, 2018-05-25:
I am flying a Mambo. Recreating of the socket does not work as far as I know. I also tried to disconnect and connect (with and without the same Mambo object).

amymcgovern, 2018-05-25:
This is an issue at the wifi level from your laptop?  If you are still *on* the wifi network, the try/catch inside the safe_send should take care of it for you.  I have seen a mode where you can't reconnect and that has been discussed lately on the parrot development forum also.  It did it to me recently at a live demo and the only solution was switching drones.  The first one finally worked again a few hours later.  Seems to be a problem inside their software as several people are reporting it on the forum.

notring, 2018-07-18:
Hello! Impossible for me to connect a bebop 2 (message "did you remember to connect your machine to the Drone's wifi network?" ) using windows, anaconda, py3.6.5, laprop wifi is connected to bebop, no remote control active...do you have any suggestion or there's a way to have a log? Thank you

amymcgovern, 2019-01-09:
this is very strange.  I haven't had the issue at all.  It sounds like a handshake problem.  Is it printing anything out?

m3m3-chan, 2019-01-10:
it gives this error "connecting connection failed: did you remember to connect your machine to the Drone's wifi network? False".  The issue is I tried it on a colleague's machine and it connected well (through windows but not the ubuntu VM) but I can't get it on to run on my machine.

amymcgovern, 2019-01-16:
been traveling and fell behind on this.  So the firewall issue sounds like it is the key.  I assume you have permission issues between the Linux VM and windows?

koyasako, 2019-01-23:
For your reference, in my env, this problem occurred due to the firewall.
I disabled ufw, and the problem is solved.

m3m3-chan, 2019-01-27:
Finally! I was able to solve the issue on windows (haven't tried Linux VM yet) Turns out it's a network configuration issue!
To solve that I had to reset the network + disable McAfee for it work. After connecting few times, the same error might return and I just reset the network adapter every time that happens. 

these are commands I used (I recommend you reset the adapter 1st before trying a whole network reset)
to reset network adapter:
`netsh winsock reset`

to reset the whole network settings:
[](https://support.microsoft.com/en-us/help/10741/windows-fix-network-connection-issues) 

I hope this is of help to anyone who suffered the same problem as me :)