amymcgovern/pyparrot #18: No module name 'Mambo'

ecamodeo, 2018-03-14:
I am not sure why I am getting this error when trying to run one of the example files.  Can anyone assist?  Typically I would pip install the missing module which I did but it did not resolve the issue.

ecamodeo, 2018-03-14:
Thanks for the quick reply.  I am somewhat new to this so I apologize for the issues.  I cloned the repo and I am running python examples/demoMamboTricks.py from the pyparrot-master folder.  Is that not correct?

amymcgovern, 2018-03-14:
No, that sounds right.  You can always change the import if it is being odd.  But Mambo is right there in the main pyparrot directory.

ecamodeo, 2018-03-14:
So I moved the demo file out of the examples directory which then resolved my import issue but created a different one, or so it seems.  I am able to find the mambo via the findMambo.py file. I changed the address and turned WiFi to False.  When running the file I now get ERROR: you are trying to use a BLE connection on a system that doesn't have BLE installed. Any thoughts?  I am using a RPI3.

amymcgovern, 2018-03-14:
you haven't installed the BLE packages as specified in the instructions.  It gives you full instructions in the wiki.

https://github.com/amymcgovern/pyparrot/wiki/Installing-pyparrot

closed 2018-03-26 by amymcgovern

LuigiCa, 2018-09-25:
Hi. I am trying to migrate from pymambo into pyparrot, but I having troubles.
I have the same problem as ecamodeo. I went through the BLE connection steps. I had some problems with the path, but I moved all the modules to target directory and solve the problem.
I seems that everything is find, but when I run the demoMamboTricks.py I got the error:
ERROR: you are trying to use a BLE connection on a system that doesn't have BLE installed. 
I would appreciate if you could share an opinion about how can I solve this issue.
regards, 
LC

amymcgovern, 2018-09-25:
Are you using linux?  BLE is only supported on Linux.

amymcgovern, 2018-09-26:
When you type python3 at the command line, can you import bluepy?

LuigiCa, 2018-09-27:
Dr. Amy McGovern,

Thank you for reviewing this issue.

I already solved the problem.

I was missing that pyparrot requires python3 instead of python, and pip3 instead of pip.

My experience with LINUX is quite limited.

Now, I am able to run the demos.

regards,

Luis Carlos Félix