amymcgovern/pyparrot #19: Not identifying Mambo.

kasinathr, 2018-03-15:
Hi All,
I am trying to Run the examples py with Pi. I installed python 3 as per documentation.
I have a Rolling spider verion 1.99.2 / HW_05
Seriel number PIO40320AE 4K160544
It has BLE interface RS_B160544 BLE
The spider works well when i use the FreeFlight Mini App how ever if i
use the code pyparrot It does not work. When i run the findMambo.py
It identify the following BLE addresses .Even if i use all of the BLE address the spider does
not fly, Is is necessary to It should be Mambo 2 or this code will work with Rolling Spider ?
See the bellow error. Any suggesion greatly appriciated.

pi@raspberrypi:~/drone/pyparrot $ sudo python3 findMambo.py
Discovered device e0:14:bc:06:3d:51
Discovered device d0:4f:7e:26:bf:31
Discovered device 68:64:4b:08:36:d2

pi@raspberrypi:~/drone/pyparrot $ python3 demoMamboTricks.py
trying to connect
 trying to connect to the mambo at address f4:f5:d8:73:8e:9b
 retrying connections
 trying to connect to the mambo at address f4:f5:d8:73:8e:9b
 retrying connections
connected: False

amymcgovern, 2018-03-15:
I do not have a spider to test so I have no idea that it will work on a spider.  All of the documentation discusses mambo and bebop because that is what I have tested.

Djsurry, 2018-03-17:
I have the same issue but with a mambo

closed 2018-03-18 by amymcgovern