amymcgovern/pyparrot #20: Python 2.7 compatibility

RobertoIV, 2018-03-17:
Hello, nice job done with this package, it has been very helpful, unfortunately I would like to ask if this can be adapted or used with python 2.7. I would like to used parrot mambo with ROS, and ROS is not compatible with Python 3.

Djsurry, 2018-03-17:
It works for me on python 2.7

amymcgovern, 2018-03-18:
It technically is working with 2.7.  The reason I went with 3 is that the threading support is much better with 3.  Also, the communication over the bytes is different between the two but I think the main place that is is an issue is fixed (at least in wifiConnection):

        try:
            # python 3
            tcp_sock.send(bytes(json_string, 'utf-8'))
        except:
            # python 2
            tcp_sock.send(json_string)

RobertoIV, 2018-03-20:
When I try to run demoMamboDirectFlight y get the following error:

`python demoMamboDirectFlight.py trying to connect
Exception in thread zeroconf-Engine:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/zeroconf.py", line 1133, in run
    reader.handle_read(socket_)
  File "/usr/local/lib/python2.7/dist-packages/zeroconf.py", line 1175, in handle_read
    msg = DNSIncoming(data)
  File "/usr/local/lib/python2.7/dist-packages/zeroconf.py", line 644, in __init__
    self.read_questions()
  File "/usr/local/lib/python2.7/dist-packages/zeroconf.py", line 667, in read_questions
    name = self.read_name()
  File "/usr/local/lib/python2.7/dist-packages/zeroconf.py", line 756, in read_name
    t = length & 0xC0
TypeError: unsupported operand type(s) for &: 'str' and 'int'

 connection failed: did you remember to connect your machine to the Drone's wifi network? 
connected: False
`
Do you know why this could be happenning?

amymcgovern, 2018-03-20:
Those errors are all in zeroconf.  I am using the python 3 version of that package and you are clearly using python 2.7.

RobertoIV, 2018-03-20:
Yes I clearly said I was using Python 2.7, that's why I wanted to know if there's a way to make it compatible with Python 2.7. It technically works on Python 2.7 for ble not for WiFi.

amymcgovern, 2018-03-20:
Ok.  I really don't know if the error isn't in my package.  zeroconf may have changed interfaces

RobertoIV, 2018-03-20:
Well, that may be the problem, if it works for you then I should downgrade the version. Thank you very much.

amymcgovern, 2018-03-20:
The error you posted was entirely in zeroconf, not in my code.  So hopefully that fixes it.

closed 2018-03-26 by amymcgovern