amymcgovern/pyparrot #21: Drone reconnecting issue

Djsurry, 2018-03-17:
So I am working on a voice-activated drone. the general gist of my code is that I use my mac to access the google speech api for voice recognition, then i use flask and requests to send the commands over to my rpi which controls the drone. Connecting at first is fine, but then when I give the command to take-off, i get the following error messages:
reconnecting to wait
trying to re-connect to the mambo at address 'my addr'
retrying connections
trying to re-connect to the mambo at address 'my addr'
It keeps doing that so my command to take off doesn't work. Any help would be appreciated!

lukask86, 2018-03-18:
Hi, I had the same issue: on Linux I disabled the firewall and at windows I used a lan-to-WiFi dongle (I think there was an issue with my WiFi driver so I tried this workaround).

amymcgovern, 2018-03-18:
I commented on this on your other issue (you opened two issues with the same error).

closed 2018-03-26 by amymcgovern