amymcgovern/pyparrot #22: Control Multiple Mambos?

stuarteve, 2018-03-21:
I am looking at some 'swarm' based work with mini-drones. I realise that the Mambo doesn't have the GPS or sensors to do proper swarming, but could the python interface be used to simulate a swarm (place the Mambos in different locations on the ground and then get each one to take of, record a photo or two and then land in sequence for example).

amymcgovern, 2018-03-21:
You can control several at once from the BLE interface (I've managed up to 7, after that the BLE bandwidth seems to run out).  With wifi, you would need a computer per mambo but shouldn't be limited on numbers (other than needing a computer each).

closed 2018-03-26 by amymcgovern

ardorem, 2018-12-12:
It has been more than 6 months.. is there any better solution than using BLE interface for this to realize the drone swarming?

amymcgovern, 2018-12-15:
It may be possible to hack the mambo wifi to give them a fixed IP address and have the all on the same router.  It is doable for the AR and I recently found instructions on doing it on the bebop 2 (though I haven't tried it yet).  I have not looked around at how to do this on the mambo but maybe the parrot developer board would know?