amymcgovern/pyparrot #23: Accelerometer

edmondja, 2018-03-21:
Hello, is there a way to gather accelerometers' information via the API ??

amymcgovern, 2018-03-21:
Mambo or Bebop?  Everything that is sent is stored.  The Mambo doesn't send that.  The bebop may?  I literally store everything sent in a dictionary for the bebop.

edmondja, 2018-03-21:
I meant the mambo. How do I know if the drone hits a wall ? I only have the camera for that ??

amymcgovern, 2018-03-22:
Unfortunately, yes.  I agree on needing the accelerometer.  I'd suggest that you post on the parrot development board and see if they have a way to get that information!

edmondja, 2018-03-22:
Technically they do, since the research paper uses an AR drone 2.0 from Parrot aswell..

amymcgovern, 2018-03-22:
As I have said, the information is inside the mambo.  You just need a way to get it out.  Post on the developer board!  I am not a parrot employee.  I don't know the internals.  Just a developer from the outside.

amymcgovern, 2018-03-22:
Also, the AR uses an entirely different SDK.  I have used it and I had access to a lot more sensors.  I have not found nearly as much information about how to get the data from the mambo.

edmondja, 2018-03-26:
Sad : 
http://forum.developer.parrot.com/t/mambo-accelerometer/7239/2
Thank you.

amymcgovern, 2018-03-26:
Sad but not surprising.  Sorry!

closed 2018-03-26 by edmondja