amymcgovern/pyparrot #24: Connection Error

ashermancinelli, 2018-03-21:
I am trying to control my Mambo drone from my python interpreter to test out the features Ill be using for my class project, but the drone keeps giving me some strange error. 

I have the drone with the camera, so the wifi should work, so I connect my laptop to the drone's wifi and give the mambo's number from the wifi hotspot's name to the Mambo constructor, and it prints to the console that ` Success in setting up the wifi network to the drone! `, before continuously printing out `timeout - trying again`, and i cannot call any of the functions on the drone object. 
Is there something I'm doing wrong?
Thanks!

amymcgovern, 2018-03-22:
The mambo's address is not needed in the constructor in wifi mode.  That is only for BLE.  

Do you have the remote controller on by any chance?  It will cause this exact behavior.

ashermancinelli, 2018-03-22:
I know the controller is off, the controller does not have any lights on. when I call the constructor without the address, it throws an error saying it expected two arguments.

ashermancinelli, 2018-03-22:
How do I initialize Mambo without giving a bluetooth address?

amymcgovern, 2018-03-22:
You don't.  I'm just saying the address is ignored if you are using wifi.

closed 2018-03-26 by amymcgovern