amymcgovern/pyparrot #25: Any OSX/Mac tutorial for PyParrot?

mulysatest, 2018-03-22:
Is there anyone managed to get PyParrot work on Mac/OSX?

amymcgovern, 2018-03-22:
It was developed and is maintained on a Mac.  What is your issue?

mulysatest, 2018-03-22:
Since the current tutorial is only for Linux (apt-get). I am now using homebrew. 
So, Is it possible to provide a detailed tutorial steps to get this code work? 
I have tried many possible way but failed to make it work especially at bluetooth package. which one I should install with brew?

amymcgovern, 2018-03-22:
Bluetooth will not work on a Mac.  It says that clearly.  The wifi works just fine though.  You don't need apt-get for any of it.  Just pip and conda, as the tutorial says.

mulysatest, 2018-03-22:
So, it mean I need to use Mambo FPV  instead of Mambo Mission?

amymcgovern, 2018-03-22:
If you want to use a Mac, yes, you need wifi.  That is because the BLE protocol is very different between Mac, windows and linux.  If there was a nice BLE package that was cross-platform, I would gladly have used it!

mulysatest, 2018-03-22:
Have you tried these package?

BLE Python Lib 
> https://github.com/adafruit/Adafruit_Python_BluefruitLE
> https://github.com/brettchien/PyBLEWrapper
> http://christopherpeplin.com/2015/10/22/pygatt-python-bluetooth-low-energy-ble/
> https://github.com/matthewelse/bleep

mulysatest, 2018-03-22:
Beside any chance I can buy just a FPV camera not the whole new set of FPV edition?

amymcgovern, 2018-03-22:
None of those are cross platform.  I chose what I chose because it is a very stable package.  If you need BLE access, Raspberry Pis are very cheap.  You may also be able to run linux in a virtual box inside the Mac (though I had issues with the permissions on that).

amymcgovern, 2018-03-22:
And yes, Parrot sells the camera.

mulysatest, 2018-03-22:
> virtual box inside the Mac (though I had issues with the permissions on that).
I tried with Vmware on Ubuntu and got same problem. 

>And yes, Parrot sells the camera.
Do you have link? All I can find is this one : https://www.parrot.com/us/drones/parrot-mambo-fpv

amymcgovern, 2018-03-22:
https://www.parrot.com/us/accessories/minidrones/camera-parrot-mambo

mulysatest, 2018-03-22:
My plan is to use it with UltraWide Band to make a swarming project. Do you open for any customer project? got any Skype ID

amymcgovern, 2018-03-22:
No, I do not do custom builds.

closed 2018-03-26 by amymcgovern