amymcgovern/pyparrot #29: Does it Possible to connect more than two mambo via BLE ?

mulysatest, 2018-03-26:
Does it Possible to connect more than two mambo via BLE and make it fly at the same time?

amymcgovern, 2018-03-26:
Yes, I have connected to up to seven at once.

mulysatest, 2018-03-26:
Do you have example code that perform that kind of task?  I need to connect to each drone in individual code: drone1.py, drone2.py .... drone7.py and execute it in parallel using in difference process right ? or I can using the same code same process?

amymcgovern, 2018-03-26:
I did it as separate processes.  I don't have lots of drones right now (my students all have all but one) but I recall there being an issue inside pybluez that prevented you from doing more than one inside the same python process.