amymcgovern/pyparrot #33: Reading IMU Values

WijdanMTak, 2018-04-07:
Hi,
Firstly, Thank you for your work, it's really nice.
secondly, I have a mambo (AIRBORNE Night drone), and I'm trying to read IMU Values while it's moving,
I've started with reading Speed_x, Speed_y, and Speed_z ... As it's shown below... but I got zeros value 
`
from Mambo import Mambo
t=0
# you will need to change this to the address of YOUR mambo
mamboAddr = "e0:14:bc:d0:3d:a1"

# make my mambo object
# remember to set True/False for the wifi depending on if you are using the wifi or the BLE to connect
mambo = Mambo(mamboAddr, use_wifi=False)

print("trying to connect")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)

if (success):

    print("sleeping")
    mambo.smart_sleep(2)
    mambo.ask_for_state_update()
    mambo.smart_sleep(2)

    print("taking off!")
    mambo.safe_takeoff(5)
    
    while(t<8):

        print("speed_x state is %s" % mambo.sensors.speed_x)
        
        
        print("speed_y state is %s" % mambo.sensors.speed_y)

        
        print("speed_z state is %s" % mambo.sensors.speed_z)

        print("battery %s" % mambo.sensors.battery)

        
        print("Flying direct: going forward (positive pitch)")
        mambo.fly_direct(roll=0, pitch=50, yaw=0, vertical_movement=0, duration=1)

        print("Flying direct: going backwards (negative pitch)")
        mambo.fly_direct(roll=0, pitch=-50, yaw=0, vertical_movement=0, duration=1)

        
        print("ask_for_state_update")
        mambo.ask_for_state_update()

        t=t+1

    print("landing")
    mambo.safe_land(5)
    mambo.smart_sleep(5)

    print("disconnect")
    mambo.disconnect()`
I thought the cause is the sequent of code,
So I have tried to make two threads, One of them to piloting and the other to update sensors and read it. but it doesn't work.
I hope BLE  give me the Possibility to read IMU Values.  
I will be grateful for any ideas.
Thank you.

amymcgovern, 2018-04-07:
They are not sending a lot of the information over wifi. That is documented in the developers forum.  They only send altitude and quaternions over wifi and not BLE.

amymcgovern, 2018-04-07:
You shouldn't have to ask for the state update in your loop either.  It comes automatically. The state update just updates the non-regular stuff (like ids)

WijdanMTak, 2018-04-07:
I see.
Thank you so much, I really appreciate that.
One more question please, Why couldn't I read the speed values, 
It's afforded via BLE, isn't it?

amymcgovern, 2018-04-07:
Are you running the latest firmware?

WijdanMTak, 2018-04-08:
I wasn't...
but I've updated it from v2.1.7 to v2.6.8, and nothing changed.

amymcgovern, 2018-04-11:
This is very odd.  I just did a test run over BLE and it is sending speed_x for me.  I'm running 3.0.17 for the firmware though so you need to update.

WijdanMTak, 2018-04-17:
Maybe because my drone is Airborne Night, I can't read these values...
I've read the release notes but nothing was mentioned...
Anyway, thank you for your help, I really appreciate that.

WijdanMTak, 2018-04-17:
I'm really sorry if I am bothering you, but it's so important for me.
Is there any way to know whether my drone Airborne Night supports reading speed or not, and so detect what's the problem... 
I apologize again.

amymcgovern, 2018-04-17:
I don't have an airborne night to try it. Sorry!

alex-kouris, 2018-04-18:
Hi, Thanks for sharing your code! :)
So if my understanding is right,, there is no way to get the altitude data through the BLE at the moment. Correct?

amymcgovern, 2018-04-18:
That is my understanding.  I saw your post on the parrot forum today also.  Hopefully one of their developers answers.

alex-kouris, 2018-04-18:
Thanks for your immediate reply. :)

maram992, 2018-04-20:
Thank you for sharing your code
I have the same problem. .. my drone Airborne night can't send the speed...
I tried ..by adding a function to mambo as the following :
def ask_for_state_update(self):
 
        command_tuple = self.command_parser.get_command_tuple("common", "Settings", "AllSettings")
        return self.drone_connection.send_noparam_command_packet_ack(command_tuple

And from the file "DroneSensorParser"
I tried to print the value in 99 &100 to notice what states the drone sends....
No speed are mentioned. ...
But there some ignoring data when the name is none ...
Any suggestions please? 
And thanks

amymcgovern, 2018-04-20:
You don't need to add that command.  It already exists!  Also, I don't think the night is supporting all of the values that you are wanting.

closed 2018-05-30 by amymcgovern