amymcgovern/pyparrot #34: ImportError: libjasper.so.1: cannot open shared object file: No such file or directory

mulysatest, 2018-04-08:
I tried to run the demoMamboVisionGUI.py but got this error msg. 

`ImportError: libcblas.so.3: cannot open shared object file: No such file or directory`

then I managed to solved it by this command 

`pip3 install opencv-python`

then I got another error when try to execute the demoMamboVisionGUI.py

`ImportError: libcblas.so.3: cannot open shared object file: No such file or directory`

then I solved it with this command 

`sudo apt-get install libatlas-base-dev`

no I got another error 
`ImportError: libjasper.so.1: cannot open shared object file: No such file or directory`

How I can solve this problem?

mulysatest, 2018-04-08:
Finally I managed to solved all dependencies 
Here it is 
```

pip3 install opencv-python
sudo apt-get install libatlas-base-dev
sudo apt-get install libjasper-dev
get install libqtgui4
sudo apt-get install python3-pyqt5
```

mulysatest, 2018-04-08:
Now I got stuck at 

`python3 demoMamboVisionGUI.py 
trying to connect to mambo now
 trying to connect to the mambo at address D0:33:78:E6:E6:11
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 
 initializing notification delegate 
connected: True
sleeping
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 3)
 Could not find sensor in list - ignoring for now.  Packet info below. 
(2, 25, 5)
Preparing to open vision
17
/home/pi/pyparrot/images
/home/pi/pyparrot/utils
Traceback (most recent call last):
  File "demoMamboVisionGUI.py", line 95, in <module>
    mamboVision.open_video()
  File "/home/pi/pyparrot/DroneVisionGUI.py", line 276, in open_video
    self.player = vlc.MediaPlayer(self.stream_adress, ":network-caching=" + str(self.network_caching))
  File "/home/pi/pyparrot/utils/vlc.py", line 3173, in __new__
    instance = get_default_instance()
  File "/home/pi/pyparrot/utils/vlc.py", line 254, in get_default_instance
    _default_instance = Instance()
  File "/home/pi/pyparrot/utils/vlc.py", line 1731, in __new__
    return libvlc_new(len(args), args)
  File "/home/pi/pyparrot/utils/vlc.py", line 4406, in libvlc_new
    ctypes.c_void_p, ctypes.c_int, ListPOINTER(ctypes.c_char_p))
  File "/home/pi/pyparrot/utils/vlc.py", line 276, in _Cfunction
    raise NameError('no function %r' % (name,))
NameError: no function 'libvlc_new'`

amymcgovern, 2018-04-08:
Sorry I haven't documented yet so I do appreciate the testing here!  Those other libraries must come with anaconda as I never had to install them.  As for VLC, you need to install the program VLC 3.0.1 or greater.  It isn't python. It's an executable available for Mac, linux, and windows.

mulysatest, 2018-04-08:
Got it, just installed VLC but now got this error, should I change to my own ip? 

> pi@raspberrypi:~/pyparrot $ cat errori2.txt 
> :~/pyparrot $ python3 demoMamboVisionGUI.py
> trying to connect to mambo now
>  trying to connect to the mambo at address D0:3A:78:E6:E6:22 
>  connected!  Asking for services and characteristics 
>  magic handshake to make the drone listen to our commandsandsensors 
>  initializing notification delegate 
> connected: True
> sleeping
>  Could not find sensor in list - ignoring for now.  Packet info below. 
> (2, 25, 3)
>  Could not find sensor in list - ignoring for now.  Packet info below. 
> (2, 25, 5)
> Preparing to open vision
> 17
> /home/pi/pyparrot/images
> /home/pi/pyparrot/utils
> [01113528] pulse audio output error: PulseAudio server connection failure: Connection refused
> [01136f10] pulse audio output error: PulseAudio server connection failure: Connection refused
> libEGL warning: DRI2: failed to authenticate
> 
> ** (python3:4090): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
> Starting user vision thread
> user callback being called
> success from play call is 0 
> [60b008a8] live555 demux error: Failed to connect with rtsp://192.168.99.1/media/stream2
> [60b02db8] access_realrtsp access error: cannot connect to 192.168.99.1:554
> [60b02db8] core access error: Connection failed
> [60b02db8] core access error: VLC could not connect to "192.168.99.1:554".
> [01329f70] core input error: open of `rtsp://192.168.99.1/media/stream2' failed
> [01329f70] core input error: Your input can't be opened
> [01329f70] core input error: VLC is unable to open the MRL 'rtsp://192.168.99.1/media/stream2'. Check the log for details.

mulysatest, 2018-04-08:
Try to change ip to my local ip 192.168.2.99 and execute command using sudo 
but still doesn't work.

[image]

amymcgovern, 2018-04-08:
Don't change your IP.  That is the IP for the mambo. I assume you have a camera on your mambo?  If so, connect to its wifi and use_wifi=true in your Mambo object creation.  It looks like you are using BLE here since it specifies an address.

mulysatest, 2018-04-08:
I see, this code only work with fpv camera. Thank you. I will try and let you know the result .

closed 2018-05-30 by amymcgovern

neurohn, 2019-01-04:
> Finally I managed to solved all dependencies
> Here it is
> 
> ```
> 
> pip3 install opencv-python
> sudo apt-get install libatlas-base-dev
> sudo apt-get install libjasper-dev
> sudo apt-get install libqtgui4
> sudo apt-get install python3-pyqt5
> ```

I needed one more step after this:
```
sudo apt-get install libqt4-test
```

amymcgovern, 2019-01-04:
Fascinating.  I definitely didn't have to do all of those but maybe they are coming with anaconda.  Thanks for posting!

VipulRamtekkar, 2019-07-03:
Hi! 
For people who are trying to install libjasper-dev on ubuntu 18.04 can do the following:
```
sudo add-apt-repository 'deb http://security.ubuntu.com/ubuntu xenial-security main'
sudo apt update
sudo apt install libjasper-dev
```

stancho-stanchev, 2020-01-06:
Guys you do not need to install any updates or other libraries. The problem is because the current version of opencv (currently 4.1.1.26) is not compatible with RPi, just run `pip3 uninstall opencv-python` and then run `pip3 install opencv-python==3.4.6.27`
This should sort you out and if you have any issues just check what versions of opencv are compatible with your Raspbian or whatever you are using.

ukBaz, 2020-01-06:
The best instructions for installing OpenCV on the Raspberry Pi can be found at:
https://blog.piwheels.org/new-opencv-builds-including-opencv-4-x/

dragosmacovei, 2020-01-16:
> Guys you do not need to install any updates or other libraries. The problem is because the current version of opencv (currently 4.1.1.26) is not compatible with RPi, just run `pip3 uninstall [...]

i tried all the versions above and some from pyimage. Running buster, from the official site. @stancho-stanchev  saved me from a lot of time waste. Thanks!

neeraj-j, 2020-02-15:
use 
`sudo apt install python3-opencv` 
instead of pip, as suggested by https://stackoverflow.com/a/60237868/9697329

lbblitz54, 2020-02-26:
> use
> `sudo apt install python3-opencv`
> instead of pip, as suggested by https://stackoverflow.com/a/60237868/9697329

Best solution

Mambela, 2020-06-13:
> use
> `sudo apt install python3-opencv`
> instead of pip, as suggested by https://stackoverflow.com/a/60237868/9697329

I have browsed around trying to install OpenCV in my RaspberryPi. 
Thank you so much for this. I have installed it successfully

ferfery001, 2020-08-12:
> use
> `sudo apt install python3-opencv`
> instead of pip, as suggested by https://stackoverflow.com/a/60237868/9697329

rockstar!! I tried for two days and this is the only solution that works for me. Thanks

motigirmaw, 2021-06-04:
how to install 
sudo apt-get install libatlas-base-dev
sudo apt-get install libjasper-dev
sudo apt-get install libqtgui4
sudo apt-get install python3-pyqt5 on windows CMD?