amymcgovern/pyparrot #35: Import Error Continues

fjaebli, 2018-04-10:
I noted the earlier post with a simliar issue but upon reinstalling and following the sequence I am continuing to get an error with demoMamboTricks,py calling Mambo.py

File "examples/demoMamboTricks.py", line 5, in <module>
    from Mambo import Mambo
ImportError: No module named 'Mambo'

If a path issue between the files, what is the best way to approach? Thank you.

Vabe7, 2018-04-10:
Did you check the file path? Since Mambo isn't a library installed by pip or so, the Mambo.py file, must be in the same folder as demoMamboTricks.py (or specified with the full path). Same principle applies to the files that get imported in Mambo.py. I don't know it for sure, but I think you should be fine by putting demoMamboTricks.py into the same folder as Mambo.py.

amymcgovern, 2018-04-10:
I need to make it a pip library :) But in the meantime, you can do what he suggested or just run it from the folder one away from examples.  So then Mambo is in that folder.  You can then do python examples/demoMamboTricks.py

s-gomezriv, 2018-04-11:
a workaround for the import errors that I found was to add 
`import sys`
`sys.path.append('../')`
before any importing in the demo files.
This extends the module search to the parent directory.

fjaebli, 2018-04-12:
running into issues with import zeroconf   Although I finally installed it, there is an import of netifaces that is now failing.  When looking, I found a source to install netifaces here:  https://pypi.python.org/pypi/netifaces

Would you recommend others?  Or do you feel this will be cover it?

It seems much of my issues stem from not having the same PI 3 environment as others when using the pyparrot code.  My students are encountering the same.  

Had to install gcc and python-dev to get zeroconf to install and get to this point.

Thanks again for the help.

amymcgovern, 2018-04-12:
That should be plenty.  I use anaconda for my default python install which already has a lot of packages.  Plus I am a developer so things like gcc are just already there by default.

fjaebli, 2018-04-13:
Quite possibly it is my lack of experience with Linux (eg Raspbian).  However, we continue to encounter issues regarding the install and modules not being seen even with os.path etc in the scripts.  I am attempting to install pyenv thinking that the version of the default Python is causing the files to be run in the wrong version of Python.    So my question is....if an install exists in one version of Python (eg 2.7) and the default version is 3.4....can installs in one version see installs in others?

amymcgovern, 2018-04-13:
No. You have to install it into the python 3 installation for it to be seen.

Vabe7, 2018-04-14:
I think that is done with sudo pip3 install instead of sudo pip install

fjaebli, 2018-04-19:
Finally resolved the issue I had with Imports failing due to modules existing in different Python installs.
Problem caused by multiple Python versions.  I already had 2 when I began the setup.

When you begin the recommended installs be sure to understand:
-  what version is associated with PIP
-  what PYTHON version is default

Good idea to ensure all installs of ZEROCONF, BLUEPY, NETIFASES are in the same PYTHON version install!

I used this link to guide some checks which ultimately lead to adding a PATH to .bashrc
[Using Python Versions](https://stackoverflow.com/questions/46820625/how-to-use-pip3-for-python-3-6-instead-of-python-3-5?rq=1&utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa)

- AND -
[Changing Defaults Python](https://stackoverflow.com/questions/46820625/how-to-use-pip3-for-python-3-6-instead-of-python-3-5?rq=1&utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa)

closed 2018-04-19 by amymcgovern