amymcgovern/pyparrot #37: How to take a picture using downfacing camera by Pyparrot

mulysatest, 2018-04-13:
I did try the MamboVision code but seems like it try to using the WIFI camera.
Does it possible to using PyParrot to take a picture using camera below the drone?

Vabe7, 2018-04-13:
Sadly interfacing the downward-facing camera isn't really possible. There is the take picture function, that utilises the downward-facing camera, but those get saved on the drone internally. Theoretically there is an option to download those pictures via a Bluetooth FTP-Server, but since this one is really slow (I think 20s per picture) it's nearly useless.

mulysatest, 2018-04-13:
yes I got that point, it's ok for me to download the file by plugin as usb. However with non-mambo fpv I can't managed to make mamboVision.py code work at all. Did you successfully run it?

Vabe7, 2018-04-13:
If you just want to take pictures you don't need DroneVision, you can just call mambo.take_picture() I think... I think that DemoMamboVision requires a Mambo FPV...

amymcgovern, 2018-04-13:
@Vabe7 is correct.  droneVision (and droneVisionGUI) are ONLY for FPVs or bebop.  The regular mambo without a camera doesn't have anything except take_picture

srijith-vijay, 2018-04-23:
the take_picture ARSDK command can be used to successfully capture the image in the drone's internal memory, using the downward facing camera. It can then be transferred using FTP via the FTP port mentioned in the response JSON packet that we receive during the initial handshake.

However, this method comes with a delay in FTP transfer.

Vabe7, 2018-04-23:
Well, I had some time in the airplane and we programmed a way to transfer the pictures from the groundcam to the pc via ftp. I just have to add some comments, than I'll create a pull-request;)

amymcgovern, 2018-04-23:
Great!

Vabe7, 2018-04-24:
Yes, at the moment it works only with WiFi from the FPV

r3dh0t, 2018-05-01:
That is awesome Vabe7. Is it fast to transfer? Can you please include an example when you commit it?

Vabe7, 2018-05-02:
I'm still working on it, you can already look into my fork and try it out, but currently I'm waiting endless for a new file on the server... Maybe on the weekend...

Vabe7, 2018-05-26:
Picture transfer is now included, I think it took about 3 seconds to grab a new image, download it and get it into opencv.

closed 2018-05-26 by amymcgovern