amymcgovern/pyparrot #39: have question about angle parsing

freaad, 2018-04-15:
Hello Captain.
I digging around about IMU data from bebop for yaw correction.
I think that the yaw for Bebop can get the data from AttitudeChanged.
Following by this information
https://github.com/AutonomyLab/bebop_autonomy/issues/158
I do not have any experience about parsing the data. could you guide the code relationship little bit to get the data?  
The class name is PilotingState I need the information from  <cmd name="AttitudeChanged" id="6" buffer="NON_ACK">
			<arg name="yaw" type="float">
				Yaw value (in radian)
			</arg>

Thank you.

amymcgovern, 2018-04-16:
All of the state information that comes from the bebop is stored in a dictionary by name.  I would print out the dictionary and see what the key is for that.  I use the name of the variable as sent by the bebop and parsed from the XML file.

freaad, 2018-04-16:
Thank you for answering.
So basically DroneSensorParser.py made the diconary right?
And then I should add the my function in bebop class.
I will try to test little bit more Thanks.

amymcgovern, 2018-04-17:
bebop.sensors.sensors_dict is a dictionary that stores ALL of the sensors.

freaad, 2018-04-17:
You already made it Thanks.

freaad, 2018-04-17:
sensors_dict is empty even if I connect successfuly.

amymcgovern, 2018-04-17:
It gets update each time sensor data comes in.  You have to be flying for some of it to be sent.

freaad, 2018-04-17:
Yes I reallize that it can update it. after I connect and start to do it.
The sensor value what I approach that
yaw=bebop.sensors.sensors_dict.get('AttitudeChanged_yaw')
roll=bebop.sensors.sensors_dict.get('AttitudeChanged_roll')
pitch=bebop.sensors.sensors_dict.get('AttitudeChanged_pitch')
I realize that yaw is very unstable (maybe magnetic variation?)
Other is very good.
Thanks.

closed 2018-04-17 by freaad