amymcgovern/pyparrot #40: pyqt problem

freaad, 2018-04-18:
Hello Captain
I am trying to use the vision data for the vision based control.
I have 2 question actually first one is pyqt5 issue.
Second one is about image object.
1.
Have you ever use this code in the windows? (DroneVisionGUI.py )
"ImportError: DLL load failed: The specified procedure could not be found"
I just use the pip install in the Anaconda virtual env.
I am trying to install the Pyqt5 and face the dll error.
I am looking for the how to fix this issue.
Does VLC require the window? In your code you said why we need to use the pyqt that it requires windows.
2. Actually this is more important to me. I am little confuse right now Please help!!!
Also I confuse that usually in the opencv example we can use the while loop to handle the every frame.
http://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_gui/py_video_display/py_video_display.html
However in your DronevisionGUI.py code and demoBebopVisionGUI.py code shows that no while loop.
Is this because of VLC  catch the latest frame already?
I just can ask the while loop and get the latest frame as cv2.imread function or below
img = bebopVision.set_user_callback_function(userVision.save_pictures, user_callback_args=None) 
This could be related with img = self.vision.get_latest_valid_picture()
however why don't you get the img object directly?

Thank you.
Best Regards
freaad

amymcgovern, 2018-04-18:
For 1:  The pyQT issue you are seeing is needing to have VLC 3.0.1 or greater installed. When the semester is a bit less crazy (e.g. over in 2 weeks!) I'll get the documentation updated and make that GUI version into an official release.

For 2: honestly, the best answer would be to avoid VLC and all of these libraries altogether and just use opencv. But it steadfastly refuses to read RTP and RTSP in a reliable way so I gave up and circumvented it.  With that said, that is why I don't follow what those docs do exactly.  You can certainly just set yourself a user call back and grab that frame.  There isn't a while loop as that happens internally inside the vision thread and then calls your user function.

freaad, 2018-04-18:
1. Ok. Thank you for your answer. Sounds like tough semester.  I just remove my virtual env and re install the everything again. I hope it is working.
2. I see. I thought that video is the continuous frame of image that is why I have to get this from start to end. However, your callback function help that I can just grab the frame when I just needed(is it right).
I will more try to dig it( I am sure I will ask again. I am very dumb ㅜㅜ.)

amymcgovern, 2018-04-18:
For 1) It is a VLC error and not python.  It needs to see the VLC libraries.  And yes there is a FAQ about that exact error and virtual env somewhere in the VLC page so doing what you just said might fix it!

2) the callback gets called as often as FPS so it is equivalent to the while loop.  It isn't continuous.  it is only as fast as the FPS of the camera.

amymcgovern, 2018-04-18:
FPS is 30 for both bebop and mambo

freaad, 2018-04-21:
It feels so bad. I only can get the 15 fps with my own code.
I try to understand your code but it is little bit hard to me.
I realize that VLC is ok but pyqt has issue with windows. 
I can run your code in the linux very smoothly.

closed 2018-04-22 by freaad