amymcgovern/pyparrot #41: No image appears on the VLC. (demoBebopVisionGUI.py)

tomoyuki1111, 2018-04-19:
I am sorry to frequently ask you a question.
I also tried to run demoBebopVisionGUI.py, and it successfully connected.
PyQt GUI appears and "Run my program" button is pushed, but there are no images on the VLC(Version 3.0.1), and there are no error codes.
I tried it and reboot  repeatedly, but no images appears. Of course, the camera cap is removed.
Please tell me the way to get the real time images correctly.

amymcgovern, 2018-04-19:
I assume you have VLC correctly installed?

freaad, 2018-04-21:
pip install python-vlc 
is required.

amymcgovern, 2018-04-21:
No.  You need the program VLC.  https://www.videolan.org/vlc/index.html

amymcgovern, 2018-04-21:
It actually installs something different than python-vlc.  That is the library.  The program is required also.

freaad, 2018-04-21:
Yes I installed both.

tomoyuki1111, 2018-04-21:
I had installed VLC from same URL you show. I don't install python-vlc.
In addition, I tried to run the sample (https://github.com/devos50/vlc-pyqt5-example) you wrote in "DroneVisionGUI.py", and It's succeeded. Maybe, VLC is installed correctly.

tomoyuki1111, 2018-05-19:
Hello
I would like to report you the success of demoBebopVisionGUI of newest pyparrot.
I don’t know the causes, but thank you very much.
I tried this in apartment with many wifi condition, and perhaps this was bad.

closed 2018-05-19 by tomoyuki1111