amymcgovern/pyparrot #43: demoBebopVisionGUI.py question

freaad, 2018-04-22:
Hello. Captain.
I hope you have good weekends.
I read your GUI example for obstacle detection for bebop.
The question is user_vision use the argument
Argument is the demo_user_code_after_vision_opened.
I do not understand the relationship between this two.
Second, I try to use the img variable with opencv.
img = self.vision.get_latest_valid_picture()
if (img is not None):
            filename = "test_image_%06d.png" % self.index
            cv2.imwrite(filename, img)
            self.index +=1
Opencv can not read it.
Do you think it is wrong approach?
Thank you.

amymcgovern, 2018-04-23:
The GUI code runs something when you click "run program".  That is what the demo is showing.

For the opencv part, I don't know your error without more info.  It is working just fine for me to output valid png images.

freaad, 2018-05-02:
Hello. I hope that you fininshed the semester.
I dig around the code and try to use it.
The demovisiongui shows the video less delay.
However when I use the my code such as like below.
movie=path+'\\utils\\bebop.sdp'#path
player=vlc.MediaPlayer(movie)
player.play()
it shows the almost 1 sec delay.
I am trying to under stand your code.
self.vision.get_latest_valid_picture()
is the 1 frame in the video is it right?
And then I can handle this as object like below?
img= bebopVision.vision.get_latest_valid_picture()
img = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
I want to do some real time image location tagging with bebop.
This delay is very noticeable to me so I want to reduce it like your code.
Thank you.

freaad, 2018-05-02:
ok So I have to dig with virtual disk. 
Could you point out that where is the image object actually is in DroneVisionGui code? I still can not run the pyqt in windows.... That is why I used my own code. I need to remove the pyqt in my code.

amymcgovern, 2018-05-03:
Do you think tkinter would work better in windows?  I could switch it to that.  I use pyqt because it was the online example from using libvlc. Since I was learning libvlc at the time, it was easier. But I honestly prefer tkinter as a python GUI development

amymcgovern, 2018-05-03:
The images are saved into a buffer in a thread that runs the function _buffer_image.

    def _buffer_vision(self):
        """
        Internal method to save valid video captures from the camera fps times a second

        :return:
        """

        # start with no new data
        self.new_frame = False

        # run forever, trying to grab the latest image
        if (self.vision_running):
            # generate a temporary file, gets deleted after usage automatically
            self.file = tempfile.NamedTemporaryFile()
            # save the current picture from the stream
            self.player.video_take_snapshot(0, self.file.name, 0, 0)
            # read the picture into opencv
            img = cv2.imread(self.file.name)

            # sometimes cv2 returns a None object so skip putting those in the array
            if (img is not None):
                # got a new image, save it to the buffer directly
                self.buffer_index += 1
                self.buffer_index %= self.buffer_size
                #print video_frame
                self.buffer[self.buffer_index] = img
                self.new_frame = True

freaad, 2018-05-03:
I try to use the pygame for GUI lol. I do not have any experience tkinter. I hope it works well.
buffer_vision get the img I still do not get it that how could I grab this img object for my obstacle avoidence.
I think I should set the networkc-caching when I start the play to reduce the speed? 
vlc.MediaPlayer(self.stream_adress, ":network-caching=" + str(self.network_caching))