amymcgovern/pyparrot #44: Getting state from "moveToChanged"

CedricMingneau, 2018-05-02:
Hello, first, thank you for the code! It's really helping me with my thesis.

I'm working with a Bebop 2 and have a question about moving the drone to GPS coordinates.
I’ve made some code that uses the “moveTo” command to make the drone reach a specific coordinate. I was wondering if I could ask the bebop whether it has reached its destination coordinate. I currently have no way of knowing when the drone actually reaches that point. Since I noticed in the code that you ask for “AllStates” (in common.xml, common class, AllStates cmd), I thought you might know how to go about this.

I’ve figured out there is a command that seems to poll for a “moveToChanged” state. Refer to the following code (in ardrone3.xml, PilotingState class, moveToChanged cmd):

```xml
<cmd name="moveToChanged" id="12">
    <comment
        title="Move to changed"
        desc="The drone moves or moved to a given location."
        support="090c:4.3.0"
        triggered="by [MoveTo](#1-0-10) or when the drone did reach the given position."/>
    <arg name="latitude" type="double">
        Latitude of the location (in degrees) to reach
    </arg>
    <arg name="longitude" type="double">
        Longitude of the location (in degrees) to reach
    </arg>
    <arg name="altitude" type="double">
        Altitude above sea level (in m) to reach
    </arg>
    <arg name="orientation_mode" type="enum">
        Orientation mode of the move to
        <enum name="NONE">
            The drone won't change its orientation
        </enum>
        <enum name="TO_TARGET">
            The drone will make a rotation to look in direction of the given location
        </enum>
        <enum name="HEADING_START">
            The drone will orientate itself to the given heading before moving to the location
        </enum>
        <enum name="HEADING_DURING">
            The drone will orientate itself to the given heading while moving to the location
        </enum>
    </arg>
    <arg name="heading" type="float">
        Heading (relative to the North in degrees).
        This value is only used if the orientation mode is HEADING_START or HEADING_DURING
    </arg>
    <arg name="status" type="enum">
        Status of the move to
        <enum name="RUNNING">
            The drone is actually flying to the given position
        </enum>
        <enum name="DONE">
            The drone has reached the target
        </enum>
        <enum name="CANCELED">
            The move to has been canceled, either by a CancelMoveTo command
            or when a disconnection appears.
        </enum>
        <enum name="ERROR">
            The move to has not been finished or started because of an error.
        </enum>
    </arg>
</cmd>
```

However, in contrast to the AllStates cmd, this cmd seems to require arguments (refer the the snippet below).

```xml
<cmd name="AllStates" id="0" timeout="RETRY">
    <comment
        title="Ask for all states"
        desc="Ask for all states.\n\n
        **Please note that you should not send this command if you are using the\n
        libARController API as this library is handling the connection process for you.**"
        support="drones"
        result="The product will trigger all states events (such as [FlyingState](#1-4-1) for the Bebop).\n
        Then, it will trigger [AllStatesEnd](#0-5-0)."/>
</cmd>
```

I tried to pass no arguments, to see what it gives. In following snippet you can see how I did this.

```python
def ask_for_move_to_status(self):
    command_tuple = self.command_parser.get_command_tuple("ardrone3", "PilotingState", "moveToChanged")
    return self.drone_connection.send_noparam_command_packet_ack(command_tuple)
```
This doesn’t really return anything. I was wondering if you know whether the “moveToChanged” cmd is a valid way for retrieving the state. If so, then should I pass parameters and what should I pass? Or how would I go about this? Any help or advice would be much appreciated! Thanks in advance.

freaad, 2018-05-02:
In my experience, you need arg about each function.

freaad, 2018-05-03:
In my previous issue shows that https://github.com/amymcgovern/pyparrot/issues/39 bebop.sensors.sensors_dict.get give the tons of current UAV status.
Also When I made the function inside of the Bebop class
#photo resoltions
    def photo_set(self,mode):
        """
        raw,jpeg,snapshot,jpeg_fisheye
        """
        (command_tuple, enum_tuple) = self.command_parser.get_command_tuple_with_enum("ardrone3",
                                                                                      "PictureSettings", "PictureFormatSelection", mode)
        return self.drone_connection.send_enum_command_packet_ack(command_tuple,enum_tuple)
#streaming resolution
    def SetResolution(self, mode="rec720_stream720"):
        #rec1080_stream480
        (command_tuple, enum_tuple) = self.command_parser.get_command_tuple_with_enum("ardrone3", "PictureSettings", "VideoResolutions",mode)

        self.drone_connection.send_enum_command_packet_ack(command_tuple, enum_tuple)
I add the argument because Arsdk ask the certain argument for the change.
Captain made the parsing part so you can do it easily.

CedricMingneau, 2018-05-16:
Sorry for the late reply. I don't always have access to the drone, so I couldn't try out your recommendations.

freaad, in your case it seems to make sense to add arguments. The drone needs to know those arguments, since it expects them for that function. It's some sort of setter method I'm guessing. In my case the method seemed more like a 'getter', so it didn't really make sense to add arguments...

I've checked out the sensors_dict. I should've looked at it earlier as it contains lots of data indeed! The dictionary contains a key called 'moveToChanged_status'. I expected this value to change when the drone reached a given coordinate. This doesn't seem to be the case.

I've also checked the 'moveToChanged_latitude' key, this value does change. In contrast to for instance 'SpeedChanged_speedX', which is a value that changes often (about every 0.2s or so), the 'moveToChanged_latitude' changed at a much slower rate. This leads me to believe this value can be used to check if Bebop has reached a given coordinate.

amymcgovern, 2018-05-16:
The sensors dictionary has everything the drone sends back!  I save it EVERY time it sends stuff.  I only made special variables for the ones I used a lot. I may well add more this summer as we are moving to a bebop project from a mambo only project.

CedricMingneau, 2018-05-16:
Yes, I noticed, it's very convenient that it works like that!

A little less on topic: is there a way to stop the drone and make it stay in its position? The 'moveTo' command makes the drone stop when it reaches its destination. I was wondering if there's a way to achieve the same effect when using the 'fly_direct' command from the Bebop-class? Currently, when I make the drone move for a couple of seconds, the drone 'drifts/slides on' until it finally comes to a stop. I'd like it to 'break' its movement, so a subsequent 'fly_direct' doesn't get 'swayed' by the 'drifting' of the previous 'fly_direct' command.

Thanks already for the quick reply!

amymcgovern, 2018-05-16:
Interesting.  If you are flying at high velocity, that is probably inevitable through physics.  The default behavior is hover.  If you are not sending any commands, it simply hovers.  But it has to be stopped first.  I haven't been flying at high velocities so I haven't had issues with it (we are working on a contest but it is indoors and I'm not flying high velocity in my house!)

freaad, 2018-05-16:
I recommend the PD control for position control.
You need to reduce the speed when you reach the goal.

CedricMingneau, 2018-05-16:
Okay Amy, so I suppose you would stop the drone by sending an inverted 'fly_direct' command? For instance:

```python
    #...
    # make it move forward
    bebop.fly_direct(0, 20, 0, 0, 2)
    # make it stop with this "inverted command"?
    bebop.fly_direct(0, -20, 0, 0, 1)
    #...
```
(I used some random pitch and duration values)

Freaad, this PD control, is it available in this project? Would you like to point me to some functions or a class I can look into? I can't immediately find anything related to PD-control...
You say you'd reduce the speed? Am I correct in assuming you would send multiple 'fly_direct' commands with smaller values? For example:

```python
    #...
    # start with a strong pitch value
    bebop.fly_direct(0, 30, 0, 0, 1)
    # reduce the speed now
    bebop.fly_direct(0, 15, 0, 0, 1)
    # reduce even more until it hovers without moving
    bebop.fly_direct(0, 5, 0, 0, 1)
    #...
```

freaad, 2018-05-16:
For the stop, you just give the
"bebop.smart_sleep(n)"
for the control, you need to define the error. 
position control, position difference is the error. When you reach the position, your error become small and you give small input. I do not know how you can calculate the x,y,z, position. 
UAV already run the pid control internally( hovering is one of the famous pid control result). Your job is that you need to give dynamic pitch, roll, yaw, throttle for position control. I also recommend that the time value around 0.1 to 0.3.

amymcgovern, 2018-05-17:
PID and PD control are used in robotics.  I haven't written it into the robots since it would be dependent on how you wanted to fly.  https://en.wikipedia.org/wiki/PID_controller

closed 2018-05-30 by amymcgovern

leocencetti, 2019-09-03:
@CedricMingneau I know it's an old issue, but did you succeed with the MoveTo command? I am trying to extend the functionality of PyParrot for a research project in my Uni, but we are struggling to get the Bebop2 to fly to a given GPS position.

CedricMingneau, 2019-09-03:
@leocencetti The MoveTo command was sufficient for my case, but it certainly was not perfect. Since the GPS inside the Bebop2 is not too accurate, the drone displayed deviations in its trajectories. These deviations worsened as more coordinates were given.

I spent some time searching for a solution or a work-around, but have not been able to find one. I suppose you force better accuracy through the application of some SLAM algorithm with cameras, but that may be out of scope for your particular use case.

Cozmo25, 2019-10-09:
> @leocencetti The MoveTo command was sufficient for my case, but it certainly was not perfect. Since the GPS inside the Bebop2 is not too accurate, the drone displayed deviations in its trajectories. [...]

Hi Cedric - could you share the code you used to send the moveTo command please?

I am sending the command to the drone (in Sphinx) and getting ack back but the drone isn't moving..

Thanks in advance

CedricMingneau, 2019-10-12:
@Cozmo25 I wrote this function (and it worked for me):
```python
def move_to_coord(self, latitude, longitude, altitude, heading):
        (command_tuple, enum_tuple) = self.command_parser.get_command_tuple_with_enum(
            "ardrone3", "Piloting", "moveTo", "NONE")
        self.drone_connection.send_coord_command(command_tuple, latitude, longitude, altitude, enum_tuple, heading)
```
First I let the drone take off (by using ```bebop.safe_takeoff()```), then I could use ```move_to_coord()``` along with some coordinates.
Please don't ask me anything specific about the code, as it has been a while since I've played around with it 😅