amymcgovern/pyparrot #45: Not really working

Psychokiller1888, 2018-05-04:
Hi!

I'm trying to use my mambo with this api that seems very promising, but unfortunately, the drone won't answer...

I'm running my script on a Raspi 3b. My program start a mqttClient that loops

When I try to use the "safe_takeoff(5)" command, it stays hanging forever on magic handshake and never takes off

> pi@snipsparrot:~/snipsparrot $ python main.py
> Loading Snips Parrot
>  trying to connect to the mambo at address e0:14:28:a2:3d:d1 
>  connected!  Asking for services and characteristics 
>  magic handshake to make the drone listen to our commandsandsensors 
>  initializing notification delegate 
> Connected to drone
>  Could not find sensor in list - ignoring for now.  Packet info below. 
> (2, 25, 3)
>  Could not find sensor in list - ignoring for now.  Packet info below. 
> (2, 25, 5)
> Snips Parrot loaded
> Taking off
>  reconnecting to send packet 
>  trying to re-connect to the mambo at address e0:14:28:a2:3d:d1 
>  connected!  Asking for services and characteristics 
>  magic handshake to make the drone listen to our commandsandsensors
> reconnecting to send packet 
>  trying to re-connect to the mambo at address e0:14:28:a2:3d:d1 
>  retrying connections 
>  trying to re-connect to the mambo at address e0:14:28:a2:3d:d1 
>  connected!  Asking for services and characteristics 
>  magic handshake to make the drone listen to our commandsandsensors

Any idea what's wrong?

amymcgovern, 2018-05-08:
That is very strange. Are you running the latest firmware?

Psychokiller1888, 2018-05-08:
Yes, I guess so, I bought it to actually try to pilot it by voice. When connected it got updated

Psychokiller1888, 2018-05-08:
I just checked the app, it says up to date, Mambo_408345 is at version 3.0.17, 0 update available

amymcgovern, 2018-05-08:
What do the eyes do?  When they are connected, they stay a solid green.  If they disconnect, they blink.

Psychokiller1888, 2018-05-08:
They stay solid green after python script connection

amymcgovern, 2018-05-08:
Strange. Can you run the demo scripts without issues?

Psychokiller1888, 2018-05-08:
When running the script directly yes. If I have any wait time between start and sending commands to it, it fails

amymcgovern, 2018-05-08:
ah ha.  are you using time.sleep() in your script?  That kills the BLE. it only works on wifi.

Psychokiller1888, 2018-05-08:
Nope... I was, then I read it kills BLE, so I went to another solution. But maybe mqtt.loop_forever() actually does sleep the thread

https://puu.sh/AixBV.png

amymcgovern, 2018-05-08:
Yes. You have to use the same solution I do inside Mambo.smart_sleep. Look at the BLE implementation if you need to make your own sleep (instead of using that call directly)

Psychokiller1888, 2018-05-08:
Ok, will check this. Thanks for pointing this out. Leaving this open to report in a couple of days

Psychokiller1888, 2018-05-09:
Well, there's no way I can change the paho-mqtt library to **not** use time.sleep(). And my mambo is not on wifi, so it's a no go for me. Thanks for your help though!

closed 2018-05-09 by amymcgovern