amymcgovern/pyparrot #49: Feature requests

amymcgovern, 2018-05-26:
GitHub doesn't run a discussion forum so I'm making this "open issue" for anyone to post feature requests to

amymcgovern, 2018-06-04:
This isn't really a feature request so make an open issue for it.  There are other users who have other drones but I don't own a cargo to test it out.

MarcoAndreettoUnitn, 2018-06-20:
Hi amymcgovern,
thank you a lot and congratulations for this beautiful work.
I am trying to develop a controller for the linear velocities of the Mambo. In practice I have a desired value of speed_x and I have to choose the corresponding pitch angle.

To develop a control algorithm I would need to have a feedback to faster rates than 2 Hz.
 
Now I can read the sensors (for instance the current speed_x) at the frequency of 2 Hz, then if I require the current value of speed_x with mambo.sensors.speed_x every 100 milliseconds, I get the same value for 5 times.

I would like to change this 2 Hz in something faster (say 20 Hz).
Is this possible?
Thank you a lot in advance

amymcgovern, 2018-06-20:
I would LOVE to be able to run the sensors at above 2Hz but this is a limitation of what Parrot is producing.  I suggest that you post on the parrot developers forum about this.  The more of us that they hear from, the more likely it is to appear in the next SDK.

MarcoAndreettoUnitn, 2018-06-21:
Got it, thank you. I will write them the issue!

Just to be sure: do the the commands have a rate of 2 Hz as well? I mean, if I write

while True:
    mambo.fly_direct(roll=0, pitch=50, yaw=0, vertical_movement=0, duration=0.1)

I assume that each iteration of the loop takes more or less 0.1 seconds, then (neglecting the communication delay) the mambo executes the action every 0.1 seconds (so faster than 2 Hz). Is this correct?

amymcgovern, 2018-06-21:
The commands run much faster than 2Hz.  It is only the sensors that are being sent at 2 Hz.  You will want to post to the forum (linked below) about that.  You will not be the only one but I'm hoping they get the message and fix it!  The bebop runs at 10 Hz (and 20 would sure be nice but 10 is far better than 2!).

http://forum.developer.parrot.com

marcosboggia, 2018-08-07:
@amymcgovern Hi, I was using your library on a parrot i got recently and its working very nice. Nice work! I would like to know if it possible to take one picture mid-air (no streaming) and then download it to my computer for example. Thanks!

amymcgovern, 2018-08-07:
You just want to save a photo from the camera?  Easy.  Just ask it for the current image at the time and then save it.

marcosboggia, 2018-08-07:
Thanks for fast reply! It is possible to make it take pictures slower? We are getting like 600 photos in just a few seconds. We only need 1 picture. Thanks

amymcgovern, 2018-08-07:
Then just save one :) You could write your program to not use the vision callback but to simply ask for the latest picture when it is that you intend to save it.  Then save it.

marcosboggia, 2018-08-07:
Ok, i tried this:
```python
userVision = UserVision(bebopVision)
#bebopVision.set_user_callback_function(userVision.save_pictures, user_callback_args=None)
success = bebopVision.open_video()
```
but the program keeps on writing lots of png images.
Where can i find the line or function that writes images to disk? 
Thanks!

amymcgovern, 2018-08-07:
The code that is saving the images is in save_pictures.  If you don't want them all, don't put a user callback and simply have your code call get_latest_valid_picture() on the vision object. Then save that picture.

marcosboggia, 2018-08-07:
This is the code we are executing, we suspect that lots of pictures are being downloaded to disk because of bebopVision.open_video(). We deleted the line that set the user callback. Also save_picture() has cv2.imwrite() commented so that function is not saving images to disk. What am i missing? Thanks!
```python
"""
Demo of the Bebop ffmpeg based vision code (basically flies around and saves out photos as it flies)

Author: Amy McGovern
"""
from pyparrot.Bebop import Bebop
from pyparrot.DroneVision import DroneVision

isAlive = False

# make my bebop object
bebop = Bebop()

# connect to the bebop
success = bebop.connect(5)

if (success):
    bebopVision = DroneVision(bebop, is_bebop=True)
    success = bebopVision.open_video()
    bebop.start_video_stream()

    if (success):
        print("Vision successfully started!")

        print("Fly me around by hand!")
        bebop.smart_sleep(5)

        print("Moving the camera using velocity")
        bebop.pan_tilt_camera_velocity(pan_velocity=0, tilt_velocity=-2, duration=4)
        bebop.smart_sleep(25)
        print("Finishing demo and stopping vision")
        bebopVision.close_video()

    bebop.disconnect()
else:
    print("Error connecting to bebop.  Retry")
```

marcosboggia, 2018-08-07:
Ohh ok, so if we want to use LibVLC is its possible with your code?

amymcgovern, 2018-08-07:
https://pyparrot.readthedocs.io/en/latest/vision.html

Spareman, 2018-10-11:
Hello there. This library has been great and I have to thank all the developers involved on that.
During working with it, I felt a lot of times the absence of this feature, but I am not sure how feasible it is to be implemented.

So, would it be possible to have a function like `go_to_position(x,y)` for the Parrot Mambo, with a reference point being the starting point of the trip? So far you can only adjust the pitch, roll and yaw of the drone limiting a lot the things that you can do with that, especially in terms of precision.

amymcgovern, 2018-10-11:
This is one of the main assignments I give out using the code to teach about the use of sensors.  So no, I will not be providing the code for it :)

marcusabate, 2018-12-02:
Hello Dr. McGovern,

Congrats on the awesome work! For state estimation it would be nice to have access to more than just velocities at every state update. Could we get accelerometer/gyroscope readings off the Mambo as well?

amymcgovern, 2018-12-03:
I only have access to what they send in the firmware.  I haven't hacked their firmware because I want this to work off the shelf.  So if they are sending it, it will show up in the sensors_dict.  Otherwise, get on the developer board for parrot and make a request.  They do answer lots of them!