amymcgovern/pyparrot #52: No sensordata when script is executed in command line

CedricMingneau, 2018-05-30:
I have a rather bizarre problem when running some of my scripts (which use the pyparrot package). Whenever I run them in PyCharm, "udp_receive_sock" receives data from the drone and sensors_dict (from Bebop) gets filled up and updates nicely. However, when I run the same script in Powershell, no data gets sent back from the drone. What's striking, is that all commands like "safe_takeoff" or "fly_direct" _do work_.

In the screenshot below I've framed the problem.

[image]

As can been seen, "udp_receive_sock" continuously times out. "batterij" is dutch for battery, and should normally display the current battery percentage.
Have I perhaps called the script in wrong way?
I've also tried to run it like this:

[image]

This gave the same effect...

Any help would be appreciated, thanks!

amymcgovern, 2018-05-30:
That is really odd!  I usually use pycharm when I am coding and running but I have certainly run from the command line before.  It seems like there is something stopping the connection and sockets.  Could there be permissions issue?  I see you are running in windows so I wonder if somehow the shell is not getting full access to permissions?

CedricMingneau, 2018-05-31:
Of course! I should've known the problem was caused by Windows Firewall. You have to disable Windows Defender Firewall for public networks. After that everything works. Thanks for the help Amy!

closed 2018-05-31 by CedricMingneau