amymcgovern/pyparrot #53: Support for other Parrot Minidrones (Cargo)

toluwabohu, 2018-06-04:
Does this work with other Parrot Minidrones? I'm trying to use it with a Cargo (Mars). 
I can find the drone:
`pi@raspberrypi:~/Workspace/drone $ sudo python3 pyparrot-master/scripts/findMambo.py 
Discovered device e0:14:1c:d3:3d:7e
`

But when running the demo i get:
`pi@raspberrypi:~/Workspace/drone $ python3 pyparrot-master/examples/demoMamboDirectFlight.py 
 ERROR: you are trying to use a BLE connection on a system that doesn't have BLE installed. 
trying to connect
connected: False
`
Of course I changed the Address in demoMamboDirectFlight.py and the error occurs as well when running as sudo.

Victor804, 2018-06-15:
Hi,
Did you follow this installation [tutorial](https://github.com/amymcgovern/pyparrot/wiki/Installing-pyparrot) ?

amymcgovern, 2018-06-15:
Documentation is now on https://pyparrot.readthedocs.io but the information is essentially the same.  I need to update the vision installation information (on my agenda for today).

petercunha, 2018-06-27:
That BLE error is caused by a failed import of the [pyparrot/networking/bleConnection.py](https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/networking/bleConnection.py) class. Many things can cause this error, but unfortunately the same error is always printed out so it's tough to diagnose. This error can even occur if you installed all the dependencies correctly.

Please comment out the try-except statement in [lines 11-15 of the Mambo.py class](https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/Mambo.py#L11-L15).

**Change this:**
```python
try:
    from pyparrot.networking.bleConnection import BLEConnection
    BLEAvailable = True
except:
    BLEAvailable = False
```

**To this:**
```python
from pyparrot.networking.bleConnection import BLEConnection
BLEAvailable = True
```

This should cause the example script to print out a more detailed error message that you should include in your next reply. It's much easier to help you with the full error message.

toluwabohu, 2018-07-01:
Dear all,
thanks a lot for your replies. I changed the lines in Mambo.py and I am getting the following error message:
`pi@raspberrypi:~/Workspace/drone $ sudo python3 pyparrot-master/examples/demoMamboDirectFlight.py`

`Traceback (most recent call last):
  File "pyparrot-master/examples/demoMamboDirectFlight.py", line 5, in <module>
    from pyparrot.Mambo import Mambo
  File "/usr/local/lib/python3.5/dist-packages/pyparrot/Mambo.py", line 16
    from pyparrot.utils.colorPrint import color_print
       ^
SyntaxError: invalid syntax
`

petercunha, 2018-07-01:
I think you may have inadvertently messed up the formatting of your Mambo.py file when you changed the try-except statement.

Can you paste what your import statements look like?

toluwabohu, 2018-07-01:
Thank you a lot. You were right. I reinstalled and for some reason now it works. Thank you a lot for the great work!

closed 2018-07-01 by toluwabohu

tianqig, 2019-09-14:
Wonder if **Parrot Minidrone is supported by PyParrot?** Come cross the discussion, from the installation guide seems Parrot MiniDrones were not listed as supported hardware.: 

Anyone can help to clarify? Thanks

“Hardware/Drone requirements¶
Parrot Mambo FPV: If you have a Mambo FPV (e.g. you have the camera), you can use the wifi interface. The wifi interface will work on Mac, Linux, or Windows.
Parrot Mambo Fly or Code: If you have a Mambo without the camera, you will use the BLE interface. pyparrot currently only supports Linux for BLE. The BLE interface was developed on a Raspberry Pi 3 Model B but it has been tested on other Linux machines.
Parrot Swing: To use the Swing you will use the BLE interface.
Parrot Bebop 2: The Bebop interface was tested on a Bebop 2 using a laptop with wifi (any wifi enabled device should work).
Parrot Bebop 1: A Bebop 1 will also work with any wifi enabled device.
Parrot Anafi: Per the development board, the Anafi should work with very minor changes. I will work to officially suppport it once the SDK from parrot is released for it.”

amymcgovern, 2019-09-14:
Minidrone is a mambo!