amymcgovern/pyparrot #54: import issue for DroneSensorParser

mmaz, 2018-06-06:
Hi @amymcgovern - thank you for providing such a friendly and educational library for Parrot drones! This package is by far my favorite option for the Mambo FPV! 

I ran into the following import error on `DroneSensorParser` after installing via `pip`; please see the associated PR which adopts the import convention from `DroneCommandParser`

```python
File "mambo/pyparrot/pyparrot/Mambo.py", line 301, in __init__
    self.sensor_parser = DroneSensorParser(drone_type="Mambo")
TypeError: 'module' object is not callable
```

amymcgovern, 2018-06-06:
Thought I had caught all of those imports associated with moving over to pip.  Thanks!  I'll accept it and push to pip also.

closed 2018-06-06 by amymcgovern

mmaz, 2018-06-06:
Thank you again!

PR #55 (2018-06-06): fix import for DroneSensorParser
fixes #54