amymcgovern/pyparrot #57: Fly the Bebop Power 2 at a certain speed / to a precise point in space

rayandaod, 2018-06-11:
Hi,
First of all thank you for your work that's really helpful.
I would like my Bebop Power 2 to take off at 2 meters from the ground.
Is there a simpler way to do that than playing with the vertical_movement and duration variables pls?
That would (I guess) also answer to my 2nd question which was how to get the drone to go to a certain 2D point in space, without changing its altitude (wondering if there is a simpler way to do that than playing with yaw, pitch, and duration variables).
Thanks a lot,
Rayan

Victor804, 2018-06-11:
Hi, I don't think it is possible to change the altitude when the drone takes off because in the documentation it is well written that the drone takes off at 1 meter from the ground. 
To answer your 2nd question you would like to use GPS coordinates ? ;)

rayandaod, 2018-06-11:
Hi! Thank you for your answer! 
Actually there is by calling the fly_direct function (bebop.fly_direct(roll=0, pitch=0, yaw=0, vertical_movement=50, duration=1) for example) but this function does not allow you to exactly enter how many meters you want the drone to take off from the ground. You have to find the right (vertical_movement, duration) pair to do so. That's why I was wondering if there was an implemented feature I did not notice.
Regarding your suggestion about the GPS, that would certainly work in an outdoor environment but I doubt it would be the case in a closed room (not enough precision I think (to move by 1 meter for example), and maybe not enough signal). Plus, the project I am working on excludes the usage of the GPS.
I appreciate your concern though :D

amymcgovern, 2018-06-12:
That I know of, you have to use the altitude.  So you do the takeoff and then you can fly direct until you get to the altitude that you want.

Victor804, 2018-06-12:
Hi,
Can we use the ultrasonic sensor data to be sure of altitude?
And as for the drone displacement I think you can use the acceleration and max speed to determine at 100 percent, how long it will take you for the desired distance.
I don't know if there's a better way to do that....

rayandaod, 2018-06-12:
I see @amymcgovern thanks. The BebopSensors class does not contain an altitude variable to update compared to the MamboSensors one. I will have to add it myself I think. 
@Victor804 yes that would be a good idea but this is not set implemented in this python wrapper for the bebop power 2, but only for the Mambo I think.

amymcgovern, 2018-06-12:
bebop does have the altitude!  It is in the sensors_dict.  Everything is in there!

rayandaod, 2018-06-13:
Just saw thank you!

closed 2018-06-13 by rayandaod