amymcgovern/pyparrot #58: Sensor data not sent by Mambo FPV drone

anoop4, 2018-06-11:
Hi,

I was working with the Mambo FPV drone connected to my laptop's wifi connection.
After connecting the drone again, the drone stopped sending the navigation data and battery status. 
In fact, it stopped sending any data for that matter.

Speed, Altitude, and battery values never change since no data is received from the drone.

However, it is accepting the control commands like takeoff, fly_direct, land etc.
I tried resetting the drone and re-flashing the firmware. Nothing seems to work. I am not sure what to do next.

Have you ever faced this issue?

amymcgovern, 2018-06-12:
No, that is very strange.  Is it happening every time?  Are you using the latest SDK?

anoop4, 2018-06-13:
Yes, this happens every time. 
I am able to see the data in Wireshark being sent from the drone to the windows computer port 43210.
However, the UDP recv_from() function times out.

amymcgovern, 2018-06-14:
This is very strange.  It sure sounds like it is a windows issue but lots of people have used windows and don't have a problem.  Do you have access to linux or Mac to try things there?

amymcgovern, 2018-06-14:
Interesting!

amymcgovern, 2019-01-09:
I fixed this bug btw.  It was a windows permissions issue.  It shows up in the FAQ now

https://pyparrot.readthedocs.io/en/latest/faq.html#i-m-using-windows-and-my-drone-gives-me-lots-of-timeout-errors

closed 2019-01-09 by amymcgovern