amymcgovern/pyparrot #59: GPS coordonate

SebArola, 2018-06-20:
Hi, I'm currently using your API to control a bebop 2 and I would like to get the gps coordonate from the drone, do you know how can I do this ? I figured out that I will have to use the command parser and the ardrone3.xml but I don't know exactly how.
Thanks, Sebastien

amymcgovern, 2018-06-20:
any information that the sensors are sending back (including the GPS info) is in the bebop.sensors.sensors_dict.  You don't need to parse it yourself.

SebArola, 2018-06-22:
Ok I'm gonna try with that, thanks.

closed 2019-01-09 by amymcgovern

roverico, 2021-03-23:
Dear Amy,
I'm dealing with the reading of bebop2 sensors from the Sphinx simulator. Finally I managed to get sensors info from the dictionary as shown below. I'd like to ask you about the meaning of the input argument 'args' within the sensor callback function (
_bebop.set_user_sensor_callback(function, args)_). Which type of optional arguments may I enter within such function? 
Many thanks in Advance, Roverico!

self.bebop.set_user_sensor_callback(self.sensors_update, None)
def sensors_update(self, args):
        # Example of sensors_dict
        # print(str(self.bebop.sensors.sensors_dict))
        #{'SpeedChanged_speedX': 0.00048239834723062813, 'SpeedChanged_speedY': 0.000755285145714879, 
        #'SpeedChanged_speedZ': -0.003092859871685505, 'AltitudeChanged_altitude': 0.9702519774436951, 
        #'AttitudeChanged_roll': -1.1307081877021119e-05, 'AttitudeChanged_pitch': -8.140489808283746e-05, 
        #'AttitudeChanged_yaw': 1.5384548902511597, 'Orientation_tilt': -13, 'Orientation_pan': 0, 
        #'MaxTiltChanged_current': 5.0, 'MaxTiltChanged_min': 5.0, 'MaxTiltChanged_max': 35.0, 
        #'NumberOfSatelliteChanged_numberOfSatellite': 12, 'GpsLocationChanged_latitude': 48.878900026261, 
        #'GpsLocationChanged_longitude': 2.3677799436439955, 'GpsLocationChanged_altitude': 1.0, 
        #'GpsLocationChanged_latitude_accuracy': 0, 'GpsLocationChanged_longitude_accuracy': 0, 
        #'GpsLocationChanged_altitude_accuracy': 0, 'PositionChanged_latitude': 48.878900026261, 
        #'PositionChanged_longitude': 2.3677799436439955, 'PositionChanged_altitude': 1.0, 
        #'MaxVerticalSpeedChanged_current': 1.0, 'MaxVerticalSpeedChanged_min': 0.5, 
        #'MaxVerticalSpeedChanged_max': 6.0, 'FlatTrimChanged': True, 'AlertSound_state': 'stopped'}
        self.msg_counter = self.msg_counter + 1
        print("Msg : "+str(self.msg_counter))
        print("Battery : "+str(self.bebop.sensors.battery))
        print("Flying state: "+str(self.bebop.sensors.flying_state))
        if "GpsLocationChanged_latitude" in self.bebop.sensors.sensors_dict:
            print("Lat: "+str(self.bebop.sensors.sensors_dict["GpsLocationChanged_latitude"]))
        if "GpsLocationChanged_longitude" in self.bebop.sensors.sensors_dict:
            print("Lon: "+str(self.bebop.sensors.sensors_dict["GpsLocationChanged_longitude"]))
        if "GpsLocationChanged_altitude" in self.bebop.sensors.sensors_dict:
            print("Alt: "+str(self.bebop.sensors.sensors_dict["GpsLocationChanged_altitude"]))

amymcgovern, 2021-04-05:
That allows you to make input arguments to your own callback functions