amymcgovern/pyparrot #63: Flat trim for Bebop

mayerjRRR, 2018-06-27:
Hi, 
I'm currently working with a Bebop v1 using pyparrot installed through pip. 
A major problem that I'm facing is that the drone will start drifting off to the side after a while. I've been previously working with [katarina](https://github.com/robotika/katarina) (switched for performance reasons) and i found out that this problem is solved by performing a flat trim on the drone once after booting. (In practice this meant, booting up the drone, connecting with katarina, issuing a flat trim command, disconnecting and connecting with pyparrot)

However, i'm missing a flat trim command in pyparrot for this purpose. I tried writing an extension method along the lines of:

    def flat_trim(self: Bebop):
        command_tuple = self.command_parser.get_command_tuple("ardrone3", "Piloting", "FlatTrim")
        self.drone_connection.send_noparam_command_packet_ack(command_tuple)

but something still seems to be wrong.
I've previously successfully experimented with sending other commands to the bebop and i'm not really doing anything differently here. 

Any help would be highly appreciated,

Cheers 
Jonas

Victor804, 2018-06-27:
Hi, 
def flat_trim(self):
    command_tuple = self.command_parser.get_command_tuple("ardrone3", "Piloting", "FlatTrim")
    self.drone_connection.send_noparam_command_packet_ack(command_tuple)

Works well for me.
Just add the function in the Bebop.py file at Bebop class level.

mayerjRRR, 2018-06-27:
Hi,
so after trying all kinds of weird hacks, i found adding `self.ask_for_state_update()` before and after above mentioned function call actually fixes the problem for me. 
No idea what is actually going on there. I have a feeling it might have to do with trying to flat trim directly after connecting and something important has not happened yet.

Victor804, 2018-06-27:
I tried this script :
```
from pyparrot.Bebop import Bebop

bebop = Bebop("Bebop")
bebop.connect(10)
bebop.flat_trim()
bebop.disconnect()
```
And the flat trim works well for me.

mayerjRRR, 2018-06-27:
Ah, i suppose it could have something to do with the fact that you're working with a Bebop 2 while i'm using the Bebop (v1).

Victor804, 2018-06-27:
I also use a bebop v1

amymcgovern, 2018-06-27:
This is perfect timing!  I'm doing demos in about 2 hours and I was trying to figure out why it was drifting.

mayerjRRR, 2018-06-27:
Glad to help!

Victor804, 2018-06-27:
@mayerjRRR Have you tested the latest version of pyparrot that should normally fix le problem?

amymcgovern, 2018-06-27:
I just demoed and it worked well :)

amymcgovern, 2018-06-27:
I am flying a Bebop 2 though.

Victor804, 2018-06-27:
On my bebop 1 it works very well too :)

closed 2018-07-14 by amymcgovern