amymcgovern/pyparrot #66: IPv4 Address Incorrect

carsonwilber, 2018-07-02:
While attempting to connect to a Bebop 2 I get the following output:

```
Service Bebop2-179788._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name=u'Bebop2-179788._arsdk-090c._udp.local.', address='\xc0\xa8*\x01', port=44444, weight=0, priority=0, server=u'Bebop2-179788.local.', properties={'{"device_id":"PI040384AH8C179788"}': False})
Traceback (most recent call last):
  File "DroneTest.py", line 7, in <module>
    success = bebop.connect(179788)
  File "C:\Python27\lib\site-packages\pyparrot-1.4.17-py2.7.egg\pyparrot\Bebop.py", line 110, in connect
    connected = self.drone_connection.connect(num_retries)
  File "C:\Python27\lib\site-packages\pyparrot-1.4.17-py2.7.egg\pyparrot\networking\wifiConnection.py", line 155, in connect
    handshake = self._handshake(num_retries)
  File "C:\Python27\lib\site-packages\pyparrot-1.4.17-py2.7.egg\pyparrot\networking\wifiConnection.py", line 294, in _handshake
    self.drone_ip = ipaddress.IPv4Address(self.connection_info.address).exploded
  File "C:\Python27\lib\site-packages\ipaddress.py", line 1333, in __init__
    self._ip = self._ip_int_from_string(addr_str)
  File "C:\Python27\lib\site-packages\ipaddress.py", line 1179, in _ip_int_from_string
    raise AddressValueError("Expected 4 octets in %r" % ip_str)
ipaddress.AddressValueError: Expected 4 octets in '\xc0\xa8*\x01'
```

It seems that wherever wifiConnection.py is getting the connection_info.address, it's coming in hex byte format with a wildcard, which is certainly not a decimal-separated octet of integers.

Not sure how I can fix this from my end. I'm using Python 2.7 and the latest pyparrot with ipaddress version 0.19.1.

amymcgovern, 2018-07-02:
This is only supported on python 3

amymcgovern, 2018-07-02:
Also, why are you giving it 179788 as the timeout for connect?

carsonwilber, 2018-07-02:
I'll be darned. Works in Python 3. I wonder if that's the only barrier to compatibility with Python 2? Everything else seems to check out...

Whoopsie, using 179788 after misreading the quick start documentation, thought I needed the number for the WiFi network to give to connect().

closed 2018-07-02 by carsonwilber