amymcgovern/pyparrot #67: I couldn't run find_mambo?

hsy77101, 2018-07-03:
I tried to run find_mambo for scanning Bluetooth Address of Mambo like the follwing.
sudo find_mambo

But I couldn't run find_mambo. There was no command.

I already installed required files by instruction of documents.
And checked documents like the following address.
https://pyparrot.readthedocs.io/en/latest/quickstartmambo.html

Where is find_mambo? 
or Do I need to run like the following?
sudo python findMambo.py

Please let me know that.
Thank you.

amymcgovern, 2018-07-03:
It should be in your anaconda/bin directory.

closed 2018-07-14 by amymcgovern

jduncan8142, 2018-08-04:
If you did a git clone in to your home directory (on linux) it is in ~/pyparrot/pyparrot/scripts

amymcgovern, 2018-08-04:
It is installed in bin if you do pip install

hsy77101, 2018-09-01:
Thank you. 
But I solved it by other way.