amymcgovern/pyparrot #69: Video recording – missing files, not storing files

machinelearner452, 2018-07-07:
I was wounding if someone could make suggestions as to where to look for the error(s) I am encounter.  I am a python newbie.

Running Windows 10, using a Mambo FPV, with VLC for video files.  I do have full control over the drone.  I am not able to store the video/images clips.  I have installed the most current VLC which displays: "Qt: Untested Windows version 10.0 detected!".  The video window does display and able to run the program for flight and landing. Using the example file "demoMamboVisionGUI.py"

There are no files stored in either of these locations:
C:\Users\Joe\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyparrot\images\
_I had to create this folder, as it was not created nor any files stored at this locations._
Or
'C:\Users\Joe\AppData\Local\Temp\tmp1mjy48rs'

The command window was logged, but not all details were recorded, the missing details included but are not complete withthis information:
…
[h264 @ 081fa300] Frame num change from 3 to 7
[h264 @ 081fa300] decode_slice_header error
[h264 @ 081fa300] Frame num change from 3 to 7
[h264 @ 081fa300] decode_slice_header error
[H264 Decoder @ 08327420] invalid mixed IDR / non IDR frames cannot be decoded in slice multithreading mode
[08229934] core video output error: Failed to open 'C:\Users\Joe\AppData\Local\Temp\tmp444bia56'
[08229934] core video output error: could not save snapshot
…

**The log file contains:**
Service Mambo_558923._arsdk-090b._udp.local. added, service info: ServiceInfo(type='_arsdk-090b._udp.local.', name='Mambo_558923._arsdk-090b._udp.local.', address=b'\xc0\xa8c\x03', port=44444, weight=0, priority=0, server='Delos.local.', properties={b'{"device_id":"PI040408AA7G558923","version":"3.0.17","rc":0}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot"}
{'status': 0, 'c2d_port': 6000, 'c2d_update_port': 51, 'c2d_user_port': 21}
c2d_port is 6000
starting listening at 
 [38;5;22m Success in setting up the wifi network to the drone!  [0m
connected: True
sleeping
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(2, 25, 3)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(2, 25, 5)
Preparing to open vision
81
C:\Users\Joe\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyparrot\images
C:\Users\Joe\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyparrot\utils
Starting user vision thread
user callback being called
in save pictures on image 0 
success from play call is 0 
in save pictures on image 0

… (many repeated lines)

in save pictures on image 0 
exiting user vision thread
landing
flying state is landed
Ending the sleep and vision
disconnecting
 [38;5;33m disconnecting  [0m

That the current details I have, thanks in advance for any suggestions to look into.

**PS a big thank you to Amy for creating and sharing all this great information.**

amymcgovern, 2018-07-07:
1) the GUI version of the vision doesn't save images out to the images directory. The regular DroneVision does.  

2) The default demoMamboVisionGUI has the image saving commented out.  It says:

            # uncomment this if you want to write out images every time you get a new one
            #cv2.imwrite(filename, img)

So you need to do that.

machinelearner452, 2018-07-08:
Amy, 
 uck you instructions were clear and I did uncomment the section before trying the saving of files:

``        if (img is not None):
            filename = "test_image_%06d.png" % self.index
            # uncomment this if you want to write out images every time you get a new one
            cv2.imwrite(filename, img)
            self.index +=1
            print(self.index)``

Anything else I should check into? Thank you

amymcgovern, 2018-07-08:
Sure hope your start of that was a typo!

The fact that it is never moving the index above 0 means it isn't seeing a valid image.  But if you are seeing a valid image stream, then that is odd.  This should be saving them into the directory where you are running your code (it doesn't add anything to the path).  

The key line that I see causing the error is:

[08229934] core video output error: Failed to open 'C:\Users\Joe\AppData\Local\Temp\tmp444bia56'
[08229934] core video output error: could not save snapshot

I'm using the tempfile package to store the snapshots.  That must be working differently in windows than on the Mac or linux.  I don't have a windows 10 machine to test it but I bet that is it given your error.  So I assume you need to ensure your machine has the right permissions for the tmp file directory.

machinelearner452, 2018-07-09:
Amy,

 Thank you for the suggestion.  I will try the file permission tomorrow.
And yes, absolutely that was a typo and oversight in my edit.  It has been
corrected in the posting. Please accept my apologies for any
possibility in communication other than my thanks to you for sharing your
great work.

Joe

machinelearner452, 2018-07-11:
Python newbie still learning and trying to get vedio and stills to work on the windows 10 laptop with a Mambo FPV. 
The currrent status is:
- Sucess with mambo FPV stills - 1) detailed below
- Issue with ground cam stills - 2) detailed below
- TBD status on FPV vedio - TBD
- On a side note I am having to run the wifi trouble shooter half the time to reconnect to the Mambo, is this an expected occurance?

1) I am now able to get the FPV pictures to record to my laptop. Will try vedio in the next day or so. 
I explored the read permission as suggested, that was part of the issue. There was a second part in that opencv2 was not working with just the VLC software installed.  **I had to also install the ffmpeg via** `pip3 install ffmpeg-python`.   Opencv2 would not work on my machine without ffmpeg - it gave a bunch of warnings with even the simplest of opencv2 operations.
_Thanks again to Amy and team for the updates of two days ago_ as it was applied!! - `7/8/2018: Version 1.4.24: switched tempfile to back to NamedTemporaryFile in DroneVisionGUI due to OS incompatibilities`

2) I am also trying to get the " demoMamboGroundcam.py " file to record stills, again with not luck, but I do get a runtime error of:
`\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyparrot\Mambo.py", line 525, in take_picture
    list = self.groundcam._get_groundcam_pictures_names()
AttributeError: 'MamboGroundcam' object has no attribute '_get_groundcam_pictures_names'`

I did try the mambo with the tablet app and was able to take a video and pictures and download from the drone to the tablet without issue.  The picture count was around a dozen on the drone, so I do not think it is a empty list issue? So there may be a real bug with that part of the code? I will contiune to share the few learnings as I come across them.

amymcgovern, 2018-07-11:
I'm so glad it is working!  I will add the ffmpeg instructions into the instructions.  I have it there for the ffmpeg approach but sounds like it is required for either VLC or ffmpeg.  

for 2) that may well be a real bug.  That code was contributed from another user who figured out how to download the ground cam images (I had it taking photos but not downloading).  Let me look at it today and I'll post an update here soon.

Vabe7, 2018-07-11:
About the second one, that's my mistake. It should be self.groundcam.get_groundcam_pictures_names() instead of self.groundcam._get_groundcam_pictures_names() in the lone where the error is reported. There is a "_" too much. You could delete that and it should work. But maybe you could try this, since I don't have a Mambo at the moment or a laptop.

amymcgovern, 2018-07-11:
Fixed!  The code is all fixed :). Upgrade to 1.4.25

amymcgovern, 2018-07-11:
It was more than just what vabe7 said.  One other more subtle bug.But both are fixed :)

machinelearner452, 2018-07-12:
downloaded the changes per  7/11/2018: Version 1.4.25 - As always thanks for the help and answers!!
Windows 10, mambo FPV running via wifi:

The groundcam works the second and third usages in that it stores a file on the mambo internal memory but does not export to a laptop file.  
The first usage with no pictures saved on the mambo has a list error in demomambogroundcam.py and cause a runtime error:
`pyparrot\demoMamboGroundcam.py", line 27, in <module>
    frame = mambo.groundcam.get_groundcam_picture(picture_names[0],True) #get frame which is the first in the array
IndexError: list index out of range
`

amymcgovern, 2018-07-12:
I'll get that fixed tonight.  Don't have my mambo with me at work today to verify any fixes but it sounds like a quick fix.

amymcgovern, 2018-07-13:
That turned out to be a quick fix.  Turns out you need to sleep a bit after taking the picture before you can ask for the name and download it.  I played a tiny bit with the length of the sleep and it seems that 0.5 seconds is enough.  I didn't do a lot of testing so it may well be able to be less than that.  Anyway, just add:

    mambo.smart_sleep(0.5)

After your take_picture() command and before everything else.

I'll fix the documentation on the demo!

amymcgovern, 2018-07-13:
tested all the way down to 0.1 and that works.

machinelearner452, 2018-07-19:
Resolution confirmed - Adding the delay removes the list error. 
As this issue is being closed, allow me to ask if anyone is aware of a means to have a video captured by a Mambo drone, mp4 or similar format?  It does not have to be live or transferable via Wifi, storing on the Mambo SD card and recovered post flight would work.  Thanks for all insight this team offers, and for any suggestions that can be provided in this area.

machinelearner452, 2018-07-19:
Closing issue

closed 2018-07-19 by machinelearner452