amymcgovern/pyparrot #71: Feature request - is video capture possible, live or post flight recoverable?

machinelearner452, 2018-07-19:
Is anyone is aware of a means to have a video captured by a Mambo drone, mp4 or similar format? It does not have to be live or transferable via Wifi, storing on the Mambo SD card and recovered post flight would work. Thanks for all insight this team offers, and for any suggestions that can be provided in this area.

amymcgovern, 2018-07-19:
If you want to do it from the FPV camera, it is easy.  Just use the images saved from the FFMPEG vision approach (DroneVision).  If you want a movie from the ground cam, the speed is slow!  But you could get about 1 frame per second and then download them all and make a movie after you land.  Just use any movie making tool.

machinelearner452, 2018-07-27:
Using DroneVision and demoMamboVision I am able to obtain about a dozen to three dozen png files before there is connection/hardware limits are reached.  The current issue is the program just continues to duplicate one of the file hundreds of times with notification that there is a lost connection.  Any suggestions on how to make more stable or at least stop recording the same image until the connection is manually broken.

On a side note as I am sure most already know, the png files can be piece together via ffmpeg -framerate 1 -i image_%03d.png -r 60 -c:v libx264 -crf 15 -pix_fmt yuv420p out.mp4

Thank you

amymcgovern, 2018-08-02:
Are you telling the video stream to stop but it keeps running?  it is multi-threaded and I can see that potentially happening, especially on windows as it is not as happy with the threads as it should be.

machinelearner452, 2018-08-05:
Yes, I am stopping the video via python command as share by this group (  mamboVision.close_video() )  The same picture file keeps repeating until I close the window command window.
Thanks as always !!

amymcgovern, 2018-08-05:
Very odd!  That is exactly the error I was seeing when I was using opencv directly, which is why I chose to use the ffpmeg or VLC approaches in the first place.

One quick question:  when it writes out to your images directory (the one in the pyparrot library), does it also do the same image over and over?  Or is that only the one in your home directory?  And that question only applies if you are using the ffmpeg approach, not the VLC approach.  The only thing I can think of with VLC may have to do with versions of both VLC and opencv.  Are you using the latest of both?

machinelearner452, 2018-08-06:
As soon as I can get the package "netifaces" working again ( I did a force reinstall as I was getting an odd error. Not a smart move on my part, the package seems to be only infrequently reviewed for issues) I will test and confirm the mambo results.  
Note: I was seeing all the working being done on the bebop, so I decided to switch the platforms last week, so I cannot recall off the top of my head without testing again.

amymcgovern, 2018-08-06:
I would suggest just a clean install for netifaces.  e.g. start over with anaconda and install again (sorry, it's ugly but it will work)

machinelearner452, 2018-08-06:
Thanks, will follow your suggestion on Tuesday, next chance to address this
project again.

machinelearner452, 2018-08-08:
Hi,  I installed anaconda 5.2, the newest for python 3.6. Then using the anaconda prompt install **Untangle** all good.   **Zeroconf** also installed successfully.  Went to try run Zeroconf.py, same issue I have been seeing.  Netifaces is not found and is not showing up on the directory list.   
Any suggestions?  I posted the issue on the netifaces site, but that site does not look to be monitored frequently. 
If the correct file structure can be shared I will try to rebuild using the downloaded files from the netifeaces site.  I can not be any worst off, I am dead in the water now.  

Before going to the anaconda path I tryied these in a different folder
pip install netifaces==0.10.7, then 6 then 5
pip install --upgrade --force-reinstall  netifaces==0.10.7,
deleting the local/pip/cache/... details 
nothing would have the netifaces.py file created in the site-package folder, which is what zeroconf.py needs to work.
There is  a **netifaces.cp36-win32.pyd** file, but not sure how/ what to do with it?  I am still learning python.

(base) c:\Users\Joe\Anaconda3\Lib\site-packages>pip install untangle
Collecting untangle
  Downloading https://files.pythonhosted.org/packages/f4/eb/a19a34858cd702b09524bd2ddb01c0d75a71f993a96aec8755a76e867911/untangle-1.1.1.tar.gz
Building wheels for collected packages: untangle
  Running setup.py bdist_wheel for untangle ... done
  Stored in directory: C:\Users\Joe\AppData\Local\pip\Cache\wheels\16\18\ff\b217dc66c822e07c274599310ba25ffce62cda02fc1c0bdca2
Successfully built untangle
Installing collected packages: untangle
Successfully installed untangle-1.1.1
(base) c:\Users\Joe\Anaconda3\Lib\site-packages>pip install zeroconf
Collecting zeroconf
  Downloading https://files.pythonhosted.org/packages/60/1f/743dd0bb96d51912a54c074729b2a264610e73018afc1ff51da3583506b3/zeroconf-0.20.0-py2.py3-none-any.whl
Collecting netifaces!=0.10.5 (from zeroconf)
  Downloading https://files.pythonhosted.org/packages/ec/8c/3c02241344b339d166ce7a8b403cee99fe423a3c200a0bc421be19bcb9af/netifaces-0.10.7-cp36-cp36m-win32.whl
Installing collected packages: netifaces, zeroconf
Successfully installed netifaces-0.10.7 zeroconf-0.20.0

results:
(base) c:\Users\Joe\Anaconda3\Lib\site-packages>zeroconf.py
Traceback (most recent call last):
  File "C:\Users\Joe\Anaconda3\Lib\site-packages\zeroconf.py", line 35, in <module>
    import netifaces
**ModuleNotFoundError: No module named 'netifaces'**

(base) c:\Users\Joe\Anaconda3\Lib\site-packages>

Thanks as always for any help/suggestions.!!

amymcgovern, 2018-08-08:
That is very odd!  I don't have windows in a 32 bit version to test but I had a bunch of students using it on windows both 32 and 63 bit just a few weeks ago and they never had to deal with that.  They would get an error when they installed zeroconf and then just did pip install netifaces (not worrying about version) and then pip install zeroconf again and it would work.

machinelearner452, 2018-08-08:
Yea, not an issue for me until an odd event made me think a force reinstall
was needed.... I hate to be this much of a rookie.

amymcgovern, 2018-08-08:
But if you just do a fresh anaconda install and no force reinstall but just a pip install netifaces, does it work?

machinelearner452, 2018-08-08:
Unfortunately no.
 I am reading the package as it looks to built as a  C  package and has to
recomplied?

amymcgovern, 2018-08-08:
On windows, that often requires you to install the c++ compiler when you install anaconda (it is an option).

amymcgovern, 2018-08-08:
Visual studio is what you want!

amymcgovern, 2018-08-08:
that is the compiler...

amymcgovern, 2018-08-08:
the steps I've done on windows:  install anaconda.  pip install pyparrot.  pip install zeroconf.  That fails because of netifaces.  pip install netifaces.  Then pip install zeroconf again.

machinelearner452, 2018-08-12:
Amy, Thank you this was very helpful!  I am back up and fly again at least one successful flight.  Using just anaconda3/conda insterface.  It may be worth noting that for those that may be on the learning curve as myself, that netifaces, is not a "netifaces. py" file, but is seen in the site-package as "netifaces-0.10.7.dist-info" directory.  Also, I did use the window 10 commands as prompt as administrator to install the files.  Not sure it is required but is the way I installed the files and I was able to reached success again.  Marking as closed!

closed 2018-08-12 by machinelearner452