amymcgovern/pyparrot #72: Wrong Python?

technologylane, 2018-07-19:
I worked through the instructions and at the end attempted to install PyParrot. I get an error that says "pyparrot requires Python '>3' but the running Python is 2.7.13" I am running the latest Raspbian Stretch. Python 2.7 isn't even an option in this release. How do I get it to see the installed Python 3?

amymcgovern, 2018-07-19:
Linux or Mac?  Make sure it is in your path. Anaconda offers to edit your .bashrc file for you automatically.

technologylane, 2018-07-19:
I'm trying to connect my Mambo with my Raspberry Pi. Using the latest Raspbian Stretch.

Victor804, 2018-07-19:
Have you tried doing `python3` to run pyparrot ?

technologylane, 2018-07-19:
Please provide a bit more information. I am not sure how to do that. I usually use Python 3 IDLE.

Victor804, 2018-07-19:
You can for example do `python3 demoMamboDirectFlight.py` in your terminal.

technologylane, 2018-07-19:
it says there is no module named 'pyparrot.Mambo'. I was thinking this was due to pyparrot not correctly installing.

Victor804, 2018-07-19:
Try moving the demo file into the pyparrot folder.
For example `mv ./examples/demoMamboDirectFlight.py ./`

technologylane, 2018-07-19:
I moved my file to the examples folder and I still have the same error message.

technologylane, 2018-07-19:
Ok... I am a step closer. It now says "No module named zeroconf". I installed this and rebooted and tried again and it still has same error.

technologylane, 2018-07-19:
OK... I think I am a step closer. It now has the error " No module named 'zeroconf'". I already installed it. I tried to install again and rebooted and have the same error.

BTW... thank you for your help so far. I teach JH/HS Computer Tech and want the students to program the drones in Python instead of Blockly. I haven't tried on my Mac. I was hoping to use with my Raspberry Pis.

Victor804, 2018-07-19:
I think you installed zeroconf on python 2. To install it on python 3 you can do: `pip3 install zeroconf`

technologylane, 2018-07-19:
We are so close. I had to reinstall a few things using pip3 instead of pip. When I run I get this error.

OpenCVAvailable is False
trying to connect
[38;5;33m trying to connect to the mambo at address D0:3A:5F:25:E6:23 [0m
[38;5;33m retrying connections [0m
[38;5;33m trying to connect to the mambo at address D0:3A:5F:25:E6:23 [0m
[38;5;33m retrying connections [0m
connected: False

Here is my code and yes the address is correct. (I am just trying to take off and land.)
"""
Demo the direct flying for the python interface

Author: Amy McGovern
"""

from pyparrot.Mambo import Mambo
from time import sleep

# you will need to change this to the address of YOUR mambo
mamboAddr = "D0:3A:5F:25:E6:23"

# make my mambo object
# remember to set True/False for the wifi depending on if you are using the wifi or the BLE to connect
mambo = Mambo(mamboAddr, use_wifi=False)

print("trying to connect")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)

if (success):
    # get the state information
    print("sleeping")
    mambo.smart_sleep(2)
    mambo.ask_for_state_update()
    mambo.smart_sleep(2)

    print("taking off!")
    mambo.safe_takeoff(5)

    sleep(5)
    
    print("landing")
    mambo.safe_land(5)
    mambo.smart_sleep(5)

    print("disconnect")
    mambo.disconnect()

Victor804, 2018-07-19:
I don't know if you did it but you should not connect your raspberry to the drone (with the graphical user interface or other). Just start the bluetooth and run your scipt.

Victor804, 2018-07-19:
Do not use the GUI to connect to the mambo. But use find_mambo.py in the /pyparrot/pyparrot/scripts/ folder. And normally you'd find your drone's address.

technologylane, 2018-07-19:
YES!!! It connected and ran the simple program to take off sleep five seconds and then land. The find_mambo script did not work, so I just tried to run my program anyway.

Thank you so much! (My dog isn't happy because the drone totally scared him!)

amymcgovern, 2018-07-20:
I'm glad!  Been offline all day so I'm super glad you got help and it worked :) and my dogs like to attack also!

Victor804, 2018-07-20:
Happy to have helped you :)

closed 2018-07-26 by amymcgovern