amymcgovern/pyparrot #73: Any way to script in interpreter?

skicombinator, 2018-07-20:
Hi.
I'm trying to use pyparrot on python3 interpreter.
I've successfully installed pyparrot and run your example code, `demoMamboDirectFlight.py`.
(Oh, my drone is Mambo Mission so it uses BLE connection and no FPV camera)
However, when I tried to script in python3 interpreter, it seems not working after making connection with drone.
Specifically speaking, I run `python3 -i init.py` in command line where `init.py` is just like following:

```python3
from pyparrot.Mambo import Mambo
mamboAddr = "some addr"
mambo = Mambo(mamboAddr, use_wifi=False)

print("trying to connect")
success = mambo.connect(num_retries=3)
print("connected: %s" % success)

if (not success):
	print("unable to connect")
	exit(1)
else:
	print("success to connect")

```

Then I got `success to connect`, so I've tried to script some simple commands like `mambo.smart_sleep(2)` or `mambo.ask_for_state_update()` or `mambo.safe_takeoff(5)` as in your example code, but it says:

```
reconnecting to send packet
trying to re-connect to the mambo at address --:--:--:--:--:--:
connected! Asking for services and characteristics
magic handshake to make the drone listen to our commandsandsensors
```

and it never returns. 
Do you have any solution for this issue?
Thanks for your help.

Victor804, 2018-07-21:
Hi, 
I tried `python3 -i demoMamboDirectFlight.py`
Personally it works well.

I use a Swing and not a Mambo but I don't think there is any difference.

It's true that when I use `ipython3` to execute commands like `safe_takeoff(10)` it returns:

```
In [4]: drone.safe_takeoff(10)
 reconnecting to send packet 
 trying to re-connect to the mambo at address e0:14:04:a7:3d:cb 
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 
 reconnecting to send packet 
 trying to re-connect to the mambo at address e0:14:04:a7:3d:cb 
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 
 reconnecting to send packet 
 trying to re-connect to the mambo at address e0:14:04:a7:3d:cb 
 connected!  Asking for services and characteristics 
 magic handshake to make the drone listen to our commandsandsensors 
```
But sometimes it works I don't know why

skicombinator, 2018-07-23:
@Victor804 Hi, thanks for the comment.
I tried `ipython` too, but still it didn't work.
In my case, no matter what I commanded with interpreter, it halts at `magic handshake to make the drone listen to our commandsandsensors`. 
Since I'm not familiar with `bluepy` or drone, I'm not even sure where to start debugging.
Had somebody have similar issue?

amymcgovern, 2018-07-23:
so it is connecting but not sending any commands?  That is very odd.  The handshake happens at the start (to tell the drone that it is a controller) and then it should be locked in.  What do the eyes do?  If the eyes (led lights) are solid green, it is still connected.  if they blink green, the connection is lost.

skicombinator, 2018-07-24:
@amymcgovern Hi, thanks for the comment.
Yeah, I think it is connecting but not sending any commands. The eyes are solid green.
For now I just write some code and execute it by `python3 code.py`, which works.

closed 2018-07-26 by amymcgovern