amymcgovern/pyparrot #79: Downloading pictures over BLE

marcosboggia, 2018-08-03:
Hi! I found out that the only way to obtain the pictures for mambo is over WiFi. I would like to know if its possible to transfer them over BLE and an idea on how that could be achieved. Thanks!

amymcgovern, 2018-08-03:
the image transfer over BLE is VERY slow.  More than 3 seconds per picture.

marcosboggia, 2018-08-03:
Thanks for the quick response! The thing is that i have to establish a connection only over BLE and obtain the pictures, I only found that possible using the android SDK forcing us to that technology. So i searched other libraries but none worked for us. Do you happen to know a way to get BLE control and picture download feature for parrot drone, available for a raspberry?

amymcgovern, 2018-08-03:
I haven't implemented the BLE protocol for downloading the photos. I looked into it but it is not documented at all and the 3 second per photo delay makes it ridiculous to use for any real-time applications.  You can take the photos over BLE and download from the phone application.

ukBaz, 2018-08-04:
Bluetooth is a collection of binary protocols for short-range wireless communication between devices.
The two main ones are:
 - Bluetooth “Classic” (also know as BR/EDR) which supports speeds up to about 24Mbps.
 - Bluetooth specification 4.0 introduced a low energy mode, “Bluetooth Low Energy” (BLE or LE, also known as “Bluetooth Smart”), that operates at 1Mbps.

I wonder if the Android SDK implements the picture functionality using Bluetooth "Classic"?

amymcgovern, 2018-08-04:
The mini drones seem to only use BLE.  Alas for speed!

ManuelCostanzo, 2018-08-04:
Hello, i really need to download pictures over ble. How can i do that ? Thanks

ManuelCostanzo, 2018-08-05:
Thanks for your reply. I have another question, there is some possibility of downloading the photo while the drone is flying ?

amymcgovern, 2018-08-05:
Over wifi, definitely.  Already works.  Over BLE?  Unsure.

marcosboggia, 2018-08-07:
Hi @amymcgovern, thanks for your reply. Currently we are trying to implement that feature in nodejs but maybe we can do it in python too, we'll se what we can do. 
Besides that i have some question about these drone. Do they all have an usable 90 degree camera pointing to the floor? (Bebop and mambo fly/mision). I couldn't find a very reliable source. Thanks!

amymcgovern, 2018-08-07:
The bebop (and I assume the anafi, though I don't have one) both have cameras that tilt.  You can point the bebop camera from 30 degrees to -90 degrees (straight down). It is also fisheye so it has a wide field of view.  Mambo does not move its camera. it is fixed forward facing.  The downward camera on the mambo is of very poor quality.

closed 2019-01-09 by amymcgovern

AlexanderDavid, 2019-10-30:
@marcosboggia Have you had any success implementing the BLE groundcam in nodejs?