amymcgovern/pyparrot #82: Wifi Connection intermittent

JackdQuinn, 2018-08-07:
Howdy,

First: Thanks!!!

Second I am encountering two problems in running the example code on our MamboFPV
1) sometimes the connection will succeed and sometimes it will time out, once timed out it seems to take a computer restart to get it to connect again.

2) safe takeoff does not seem to work, the drone crashes.
Does anyone have a very simple takeoff /land script I can start my students on?

-Cheers,
Jack

amymcgovern, 2018-08-07:
Thanks!

For:

1) It can timeout if it never succeeded in disconnecting on the first try.  Alas, rebooting is the only solution I know for that because the connection creates a socket and the socket has to be released through disconnect.

2) How does it crash?  Sounds like an issue with the drone.  Does it work with RC?  Because the only time I see crashes on takeoff are when a motor or propeller is broken.

machinelearner452, 2018-08-07:
Jack, 
I second your thank you to Amy and the team !

 I too was playing with the mambo for a while and had connected issues as well.  I ended up making a batch file for windows to connect and enddd up making a batch file.  You may not want the logging function so remove **>> wifi.txt**, also replace the x's with you wifi ID. The batch can be upgrade by checking if the IP address is good, then there is no need to run the balance of the wifi script as it take about 3-5 mins of battery time away. 

`del wifi.txt

netsh wlan connect name="Mambo_xxxxxx" >> wifi.txt

netsh winsock reset >> wifi.txt

netsh int ip reset >> wifi.txt

ipconfig /release >> wifi.txt

ipconfig /renew >> wifi.txt

netsh wlan connect name="Mambo_xxxxxx" >> wifi.txt`

JackdQuinn, 2018-08-08:
Hi ML,

Like the batch file but still getting connection timeouts once I run the py scripts.
[image]

machinelearner452, 2018-08-12:
JackdQuinn, 
 if you are still having an issue, may I suggest you download a wifi analyzer on your phone and check to see if you have a bunch of background noise going on with the wifi channel you are using?  Second, check your wifi setup to ensure you have it as stay connected. otherwise, it may disconnect and not reconnect automatically.

JackdQuinn, 2018-08-16:
Thanks for all the help!!! It looks like it was an issue with a particular drone. Once I switched to others everything worked fine.

closed 2018-08-16 by JackdQuinn