amymcgovern/pyparrot #84: Be

hassanmjawhar, 2018-08-14:
`"""
Demo the Bebop indoors (sets small speeds and then flies just a small amount)
Note, the bebop will hurt your furniture if it hits it.  Even though this is a very small
amount of flying, be sure you are doing this in an open area and are prepared to catch!

Author: Amy McGovern
"""

from pyparrot.Bebop import Bebop

bebop = Bebop(drone_type="Bebop2")

print("connecting")
success = bebop.connect(10)
print(success)

if (success):
    print("turning on the video")
    bebop.start_video_stream()

    print("sleeping")
    bebop.smart_sleep(2)

    bebop.safe_takeoff(10)
    bebop.smart_sleep(5)
    bebop.safe_land(10)

    print("DONE - disconnecting")
    bebop.stop_video_stream()
    bebop.smart_sleep(5)
    print(bebop.sensors.battery)
    bebop.disconnect()`

The bebop2 drone is starting normal, flying and landing safe:
It's throwing an error and not starting video stream:

Logs:
`c2d_port is 54321
starting listening at
 Success in setting up the wifi network to the drone!
True
turning on the video
sleeping
 Could not find sensor in list - ignoring for now.  Packet info below.
(144, 0, 1)
 Could not find sensor in list - ignoring for now.  Packet info below.
(134, 0, 3)
 trying to land
 Could not find sensor in list - ignoring for now.  Packet info below.
(144, 0, 1)
 Could not find sensor in list - ignoring for now.  Packet info below.
(134, 0, 3)
DONE - disconnecting
24
 disconnecting`

Any help is much appreciated.

amymcgovern, 2018-08-14:
Those errors are not really errors.  They are just telling you about some sensors that I ignore (because they are not in the published files).  As for the video, you need to look at the docs where it talks about how to run video. 

https://pyparrot.readthedocs.io

The video stream is indeed starting.  You just are not displaying it.

hassanmjawhar, 2018-08-16:
Alright, thank you.

closed 2018-08-16 by hassanmjawhar