amymcgovern/pyparrot #86: Newbie - keyboard interface attempt

machinelearner452, 2018-08-17:
Hi,  If any has any time to offer feedback or a more effective way to create a keyboard interface to the bebop2, I would appreciate the tips. 
Able to get connected to the drone, but the flight controls are not working and getting a clear indicator that I am creating more than one wifi scoket connect.  I.e., my logic is flawed - still much for me to learn. 
I dd it this way as the 'pynput" package would not release control (until the return was false), and I thought to keep the keyboard interface as the managing application was a reasonable solution....
Thanks as always for any time anyone can offer.

Setup = Windows 10, python 3.6 bebop2

The screen displays are: 
```
C:\Users\Joe\Anaconda3\Lib\site-packages\pyparrot>keyboard_dc_2.py
"Zeroconf called and running - successfully - print statement added as a check on Zeroconf"
"This is keyboard interface to the Parrot Bebop drone."
"The prep steps are the wifi needs to be turn on and the connect established"
connecting
Service Bebop2-176199._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-176199._arsdk-090c._udp.local.', address=deleted', port=44444, weight=0, priority=0, server='Bebop2-176199.local.', properties='deleted': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at
 [38;5;22m Success in setting up the wifi network to the drone!  [0m
True
**keyboard enter: q**
 bebop.ask_for_state_update()
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(144, 0, 2)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(142, 0, 3)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 3)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Error: tried to parse a bad sensor packet  [0m
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(144, 0, 1)
**other keys checked**
bebop.safe_takeoff(10)
bebop.ask_for_state_update()
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(144, 0, 2)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(142, 0, 3)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 3)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(134, 0, 4)
 [38;5;196m Error: tried to parse a bad sensor packet  [0m
 [38;5;196m Could not find sensor in list - ignoring for now.  Packet info below.  [0m
(144, 0, 1)
**more key checks:**
bebop.safe_land(10)
bebop.move_relative(0, 0, -0.1, math.radians(0))
bebop.move_relative(0.1, 0, 0, math.radians(0))
bebop.move_relative(-0.1, 0, 0, math.radians(0))
bebop.move_relative(0, 0, 0.1, math.radians(0))
bebop.move_relative(0, 0, -0.1, math.radians(0))
bebop.safe_takeoff(10)
bebop.move_relative(0, 0, 0, math.radians(90))
bebop.move_relative(0, 0, 0, math.radians(-90))
bebop.start_video_stream()
bebop.stop_video_stream()
bebop.emergency_land()
This is keyboard interface to the Parrot Bebop drone.
[... 121 lines cut ...]
        print("bebop.move_relative(0, 0, 0.1, math.radians(0))")
        bebop.move_relative(0, 0, 0.1, math.radians(0))
        return 
	
    elif key == keyboard.Key.left:
        # Send drone left 0.1 meters
        print("bebop.move_relative(-0.1, 0, 0, math.radians(0))")
        bebop.move_relative(-0.1, 0, 0, math.radians(0))
        return
    
    elif key == keyboard.Key.right:
        # Send drone right 0.1 meters
        print("bebop.move_relative(0.1, 0, 0, math.radians(0))")
        bebop.move_relative(0.1, 0, 0, math.radians(0))
        return
				
		
		
# Collect events until released
with keyboard.Listener(
        on_press=on_press,
        on_release=on_release) as listener:
    listener.join()
	
````

amymcgovern, 2018-08-17:
you are only calling connect once so that is odd...

Victor804, 2018-08-23:
Hi,
I'm working on a script that allows me to control the parrot swing with a joystick. It's pretty simple to modify it for bebop I think (I'm not trying yet). 

Here is the link if you are interested: https://github.com/Victor804/pyparrot/blob/joystick_support/examples/demoSwingJoystick.py

machinelearner452, 2018-08-23:
Victor804,
 Thank you for sharing.  I will look it over.  I was able to get the
keyboard interface working somewhat. The drone takes flight commands but
does time out unexpectedly. I need more testing over the wekkend to better
define the issue I am seeing.

 I will mark this issued closed as soon as the interface is stable and get
the video feed working.
Thanks again!

machinelearner452, 2018-08-29:
The issue I was calling "zeroconf" a second time via the "import pyparrot.keyboard_dc" code. Thus newbie...
The keyboard functions are working OK, but the drone does not seem to be very repeatable. I say this as typing the same command in the same order puts the drone in different locations, off by about a foot on a few tests. Has anyone else tried a repeatability test? Lastly, can not get the video to work with any repeatability or for any length> 2120 seconds.  Will open a new issue to get some suggestions. Thanks as always for the help and insight.

closed 2018-08-29 by machinelearner452

amymcgovern, 2018-08-29:
are you using a mambo or a bebop?  Mambo sensors are only at 2Hz.

machinelearner452, 2018-08-29:
Mambo.    So you are saying that there is no hope to get a reason png set of pictures or a live video feed from the mambo?

If so, how is the Mambo android interface able to have FPV flight control while supporting video storage on the SD card?  Do we have the same access (bit rate data transfer) as available to the Android app via the access parrot allows via the XLMs you have nicely created and shared? If that is the case, no issue I will switch to the Bebop 2, and break out the paint for all the walls I will be hitting and denting...
Thank you!

amymcgovern, 2018-08-29:
I get a live video feed easily from the front camera.  It is about .25 to .5 second behind (which does annoy me, I wish I could get it to be fully real time!) but it is live and works just fine long-term for me.  

You can store photos on the SD card if you are using the downward facing camera.  If you want to download from that camera, it seems to run about 0.5 second behind between sending the "take photo" command and then being able to download it.

machinelearner452, 2018-08-29:
That is good news, - when I learn a little more and reach that point, I too can deal with a lag of .25 to .5, it just means slower flight speed and less flight time, as you know. 

Are you aware of any way to have the Mambo plugged in and charging while still being able to have the wifi connection? I would not think so but wanted to ask.

Lastly, In the morning I may open an issue/question around release 1.5.1 and demoMamboVisionGUI.py example not working correctly with Windows 10.  I want to test again with a fresh battery in the AM before wasting anyone time.

Amy, thank you again for all you do and offer your user group !!!

amymcgovern, 2018-08-29:
Oh having it plugged in and working would be so nice!  Not that I know of :(

For the windows, I'll do my best to fix.  I don't have a windows machine myself (I guess I need to do that) but I tested a bunch of stuff with one of my students and we fixed the permissions issues.  What works on Mac and linux was misbehaving on windows so I just hard-coded the file name.  But I'll be happy to help again.

machinelearner452, 2018-08-29:
Maybe I will try to hack a battery apart and solder in a USB cable? Not pretty, but functional.

Let me confirm before I waste any of your time.

machinelearner452, 2018-08-31:
Amy, May I ask if you would be willing to share the code you created to have the live streaming video on a mambo?  I ask as have 3 mambos, and am getting different results every time I run the same code even with the same mambo. The video capture will work one run and not the next.  My goal is to allow my wife to use the examples in her fifth-grade classroom. The mambo is a little safer for classroom vs the bebop 2. Thanks either way.  Joe - email is joelarson4524524@gmail.com

amymcgovern, 2018-08-31:
Which code are you looking for that is different than what is in the demos?  What specifically do you want it to do?

machinelearner452, 2018-08-31:
I am trying to capture the rstp link real time into a mp4 file.  Half the
time the rstp display is blacked out with nothing saved, but my code with
the  keyboard commands is working all the time. I am guessing I have
created a conflict or race condition some where?

My goal is to capture the rstp output to a mp4 file. From there add in a
key board command function.

amymcgovern, 2018-08-31:
Ok, so you are trying to do it using opencv opening the RTSP directly?  I cannot make that work reliably, that is why I wrote the other two approaches (using VLC and FFMPEG).  It *should* just be one line of code to open the RTSP library in opencv but it fails quite reliably.  

Does the GUI version of the vision not work for you?  It shows a live (with about 0.5 sec delay) window.