amymcgovern/pyparrot #91: Camera tilt not working

ManuelCostanzo, 2018-08-29:
Hello ! I have a problem with camera tilt. I want to show the streaming as 90 degree to floor.

I use this command: bebop.pan_tilt_camera(90.,90.) but not working.

Any solutions ? Thank you.

marcosboggia, 2018-08-29:
Hi!, I am having same issue. Tested multiple combinations of number and pan_tilt_camera()/pan_tilt_camera_velocity() calls, even changing command OrientationV2 to Orientation with no luck. Thanks!

Victor804, 2018-08-29:
Hi,
Have you read the comments of the functions you use ? 

[pan_tilt_camera](https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/Bebop.py#L497)
[pan_tilt_camera_velocity](https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/Bebop.py#L513)

I think it will help you.

marcosboggia, 2018-08-29:
Yes, I tried setting different velocities, durations with low and high values several times and the camera never moved.

amymcgovern, 2018-08-29:
I use 

bebop.pan_tilt_camera_velocity(pan_velocity=0, tilt_velocity=-10, duration=1)

Inside a while loop that watches the actual orientation achieved.

I am at work without my drone (alas, not allowed to fly the bebop on campus!) so I can't test but that worked last time I ran my code!

amymcgovern, 2018-08-29:
p.s are you all on bebop 1 or 2?  I will re-test on my bebop 2 when I get home.

marcosboggia, 2018-08-29:
Oh sorry, I am using Bebop 1. I will try that in a few moments. Thanks!

marcosboggia, 2018-08-29:
Tested this code:
```python
bebop = Bebop('Bebop')
success = bebop.connect(5)
bebop.start_video_stream()
while True:
    bebop.pan_tilt_camera_velocity(pan_velocity=0, tilt_velocity=-10, duration=1)
```
and camera didn't move a bit. It keeps looking straight forward. Also tried different tilt_velocity values.

amymcgovern, 2018-08-29:
I'm going to guess the bebop 1 has different commands.  I don't have a 1 to test on!  @Victor804 you have a 1 also, right?

Victor804, 2018-08-29:
Yes, I have a bebop 1. I will try.

Victor804, 2018-08-29:
I confirm it does not work on my bebop 1

Victor804, 2018-08-29:
I finally managed to use pan_tilt_camera by modifying it slightly. 
But @amymcgovern it works with a bebop 2 ?

amymcgovern, 2018-08-29:
Yeah.  What did you modify?  Maybe we can make a version that works for both?  It's hard since I don't have a 1 to test.  I will verify current working status on my 2 when I get home.  I haven't used it on the bebop 2 since release 1.5.1 (I was using 1.4.26).

marcosboggia, 2018-08-29:
What changes did you have to do to make pan/tilt function work for bebop 1? Thanks

Victor804, 2018-08-29:
I use the command ardrone3/Camera/Orientation and not ardrone3/Camera/OrientationV2. 
Normally Orientation is obsolete...
Here's the code I used:
```
    def pan_tilt_camera(self, tilt_degrees, pan_degrees):
        """
        Send the command to pan/tilt the camera by the specified number of degrees in pan/tilt

        Note, this only seems to work in small increments.  Use pan_tilt_velocity to get the camera to look
        straight downward

        :param tilt_degrees: tilt degrees
        :param pan_degrees: pan degrees
        :return:
        """
        command_tuple = `self.command_parser.get_command_tuple("ardrone3",` "Camera", "Orientation")

        self.drone_connection.send_param_command_packet(command_tuple, param_tuple=[tilt_degrees, pan_degrees],
                                                        param_type_tuple=['i8', 'i8'], ack=False)
```

Victor804, 2018-08-29:
This is the program I used to try:
```
"""
Demo of the Bebop vision using DroneVisionGUI (relies on libVLC).  It is a different
multi-threaded approach than DroneVision

Author: Amy McGovern
"""
from pyparrot.Bebop import Bebop
from pyparrot.DroneVisionGUI import DroneVisionGUI
import threading
import cv2
import time

isAlive = False

class UserVision:
    def __init__(self, vision):
        self.index = 0
        self.vision = vision

    def save_pictures(self, args):
        #print("saving picture")
        img = self.vision.get_latest_valid_picture()

        if (img is not None):
            filename = "test_image_%06d.png" % self.index
            #cv2.imwrite(filename, img)
            self.index +=1

def demo_user_code_after_vision_opened(bebopVision, args):
    bebop = args[0]

    print("Vision successfully started!")
    #removed the user call to this function (it now happens in open_video())
    #bebopVision.start_video_buffering()

    # takeoff
    # bebop.safe_takeoff(5)

    # skipping actually flying for safety purposes indoors - if you want
    # different pictures, move the bebop around by hand
    #print("Fly me around by hand!")
    #bebop.smart_sleep(15)

    if (bebopVision.vision_running):
        print("Center camera")
        bebop.pan_tilt_camera(0, 0)
        bebop.smart_sleep(5)
        print("Moving the camera using velocity")
        bebop.pan_tilt_camera(-90, 0)
        bebop.smart_sleep(5)
        print(bebop.sensors.sensors_dict)
        bebop.smart_sleep(5)

        # land
        #bebop.safe_land(5)

        print("Finishing demo and stopping vision")
        bebopVision.close_video()

    # disconnect nicely so we don't need a reboot
    print("disconnecting")
    bebop.disconnect()

if __name__ == "__main__":
    # make my bebop object
    bebop = Bebop("Bebop")

    # connect to the bebop
    success = bebop.connect(5)

    if (success):
        # start up the video
        bebopVision = DroneVisionGUI(bebop, is_bebop=True, user_code_to_run=demo_user_code_after_vision_opened,
                                     user_args=(bebop, ))

        userVision = UserVision(bebopVision)
        bebopVision.set_user_callback_function(userVision.save_pictures, user_callback_args=None)
        bebopVision.open_video()

    else:
        print("Error connecting to bebop.  Retry")

```

amymcgovern, 2018-08-29:
That's interesting.  OrientationV2 is definitely what you need on the Bebop 2.  But that would be easily fixable in general since it knows that bebop it is during initialization.

Victor804, 2018-08-29:
What I find strange is that OrientationV2 should be compatible with bebop 1 and bebop 2 https://developer.parrot.com/docs/reference/bebop/index.html#move-the-camera-deprecated

As for pan_tilt_camera_velocity, I don't think this command will work with bebop 1, but I think we can simulate it with pan_tilt_camera.

amymcgovern, 2018-08-30:
Thanks to @Victor804 for testing on the bebop 1.  There is now a version 1.5.2 that checks for what kind of bebop you have and uses the older command for the bebop 1.  Should solve your problem.

ManuelCostanzo, 2018-09-01:
Thanks to all !

closed 2019-01-09 by amymcgovern