amymcgovern/pyparrot #93: Please how to solve "NameError: name 'DefaultDelegate' is not defined" for BLE.

hsy77101, 2018-09-01:
My environment is  Raspberry Pi 3 Model B, preinstalled  python3

I am trying Parrot Mambo with BLE connection.
So I installed all of your mentioned in "https://pyparrot.readthedocs.io".

After then I tried to scan by using following link.
"https://pyparrot.readthedocs.io/en/latest/quickstartminidrone.html"

But I met the following error. I don't know how I solve this error.
Please let me know how to solve it. Thank you.

pi@raspberrypi:~/pyparrot/pyparrot/scripts $ sudo python ./findMinidrone.py
Traceback (most recent call last):
  File "./findMinidrone.py", line 18, in <module>
    class ScanDelegate(DefaultDelegate):
NameError: name 'DefaultDelegate' is not defined

amymcgovern, 2018-09-01:
Did you install the bluepy package?

hsy77101, 2018-09-01:
Yes, I did. 
I solved this issue. 
I just run "pi@raspberrypi:~/pyparrot/pyparrot/scripts $ sudo python3 ./findMinidrone.py"
instead of "pi@raspberrypi:~/pyparrot/pyparrot/scripts $ sudo python ./findMinidrone.py".
I just replace "python" to "python3" of command.

I think you need to update document with exact information.
Anyway, Thank you for your comment.

amymcgovern, 2018-09-01:
Can't exactly write it when people have all different python installations.  You just need to know to run python 3.  But I will update it into the FAQ to ensure people are running python 3 and knowing the python 2 error helps!

hsy77101, 2018-09-02:
Ok. Dr.Amy. Thank you.
Your pyparrot is great for us.

closed 2018-09-02 by hsy77101