amymcgovern/pyparrot #95: Is "demoMamboGroundcam.py" for only mambo FPV?

hsy77101, 2018-09-02:
Hello.

I am trying to test "demoMamboGroundcam.py ".
As you know, there is no WiFi function for mambo fly, mambo code, mambo mission except mambo fpv.
Can you answer about the following question?

1. demoMamboGroundcam.py  is for Mambo FPV?
2. Do you mean that Groundcam is camera installed bottom of Mambo? or External attachable camera like Mambo FPV wifi Camera?

amymcgovern, 2018-09-02:
For your questions:

1) Yes, it only works for Mambo FPV.

2) There is a ground cam in all of the mambo's.  But the protocol to get the data off it using BLE is not implemented in pyparrot because it isn't documented at all by parrot.  So you can take pictures using the ground cam in pyparrot but you can't download except via the phone app IF you only have BLE.  If you have the FPV (and thus the wifi), you can use the ground cam demo methods to download.

hsy77101, 2018-09-02:
Dear Dr. Amy

Thank you for answer kindly.
Always your comments is very helpful for me.
Actually, I am making easy document how to use mini drone by python with Raspberry Pi for students.
It will be nice job. 
Actually, in our country, so many students don't understand documents in English well even though their parents's too much investment for learning English.

closed 2018-09-02 by hsy77101