amymcgovern/pyparrot #97: Parrot Bebop2 connect issue

LeongJunCheng, 2018-09-03:
when i type the "success = bebop.connect(10)"
it will come out these:

connecting
Service Bebop2-A391765._arsdk-090c._udp.local. added, service info: ServiceInfo(type='_arsdk-090c._udp.local.', name='Bebop2-A391765._arsdk-090c._udp.local.', address=b'\xc0\xa8*\x01', port=44444, weight=0, priority=0, server='Bebop2-A391765.local.', properties={b'{"device_id":"PI040376AA6A391765"}': False})
{"d2c_port": 43210, "controller_type": "computer", "controller_name": "pyparrot", "arstream2_client_stream_port": 55004, "arstream2_client_control_port": 55005}
{'status': 0, 'c2d_port': 54321, 'c2d_update_port': 51, 'c2d_user_port': 21, 'qos_mode': 0, 'arstream2_server_stream_port': 5004, 'arstream2_server_control_port': 5005}
c2d_port is 54321
starting listening at [38;5;22m Success in setting up the wifi network to the drone! [0m

timeout - trying again
timeout - trying again
timeout - trying again
timeout - trying again
timeout - trying again

How to fix the timeout - trying again?

amymcgovern, 2018-09-03:
Windows?  Do you have user input in there?

LeongJunCheng, 2018-09-04:
yup Windows, i solve it by closing the Firewall.

closed 2019-01-09 by amymcgovern