amymcgovern/pyparrot #98: Dependency on Bluepy

ukBaz, 2018-09-03:
Hello,

I've been following this project for a while as it looks great. I have been watching the repository as I have an interest in using Bluetooth Low Energy (BLE) with students. I saw that there was a dependancy on Bluepy and wondered if there was any interest for the project not to have this dependancy?

Since Bluepy was conceived the core BlueZ project has done work to better support interfacing with different programming languages with an API based on DBus.
For this project the key pieces of the BlueZ API would be:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt

There is then a very nice Python library for interfacing with DBus on Linux:
https://github.com/LEW21/pydbus
This can be installed with `sudo pip3 install pydbus`

One of the less straight forward parts of this approach is converting BLE UUID's to DBus objects.
I did a presentation on this at my local Python User Group which can be seen here:
https://github.com/campug/bzero_kata/blob/master/PythonHandsOn.pdf

If I've understood this library correctly the only place that Bluepy is imported is in:
https://github.com/amymcgovern/pyparrot/blob/master/pyparrot/networking/bleConnection.py
which should make it more straight forward to change the dependancy.

The other place Bluepy is used is `findMambo.py` script. I took a go at moving this to the BlueZ DBus API to give an example.

```python
"""
Find the BLE address for a mambo.  To run this,

python findMambo.py

sudo is not required for this.
"""
from pydbus import SystemBus
from gi.repository import GLib

class Scanner:
    def __init__(self):
        self.dbus = SystemBus()
        self.adapter = self.dbus.get('org.bluez', '/org/bluez/hci0')
        self.bluez = self.dbus.get('org.bluez', '/')
        self.loop = GLib.MainLoop()
        self.bluez.onInterfacesAdded = self.handle_discovery

    def scan(self, duration):
        GLib.timeout_add_seconds(duration, self.timeout)
        self.adapter.StartDiscovery()
        try:
            print('start discovery...')
            self.loop.run()
        except KeyboardInterrupt:
            self.timeout()

    @staticmethod
    def handle_discovery(path, interface):
        new_device = None
        if 'org.bluez.Device1' in interface:
            new_device = interface['org.bluez.Device1']
        if 'Name' in new_device:
            if 'Mambo' in new_device['Name']:
                print('Found a Mambo')
                print('Device {}, RSSI={} dB'.format(new_device['Address'],
                                                     new_device['RSSI']))

    def timeout(self):
        print('Stopping discovery')
        self.adapter.StopDiscovery()
        self.loop.quit()

def main():
    scanner = Scanner()
    scanner.scan(10)

if __name__ == "__main__":
    main()
```

After taking a look at `bleConnection.py` and at https://forum.developer.parrot.com/t/minidrone-characteristics-uuid/4686/3 it appears it might not be a trivial task. However it might prove more straight forward for someone that already has an understanding of these.

Either way I thought it was worth starting a discussion to see if it was something that was a priority to the project.

amymcgovern, 2018-09-03:
When I wrote the library, the wifi option did not exist.  I am much more focused on the wifi support now because the sensors are much better (more information and you can get the video).

ukBaz, 2018-09-03:
Thanks for the quick response. Sounds like what you have is great for your requirements.

closed 2018-09-03 by ukBaz

amymcgovern, 2018-09-03:
Yeah.  If someone wants to contribute other code using a different BLE approach, I'm fine with it!  But I'm focused on the wifi for what I'm working on.