amymcgovern/pyparrot #99: Indoor flight with Bebop1

ManuelCostanzo, 2018-09-04:
Hello, I am testing bebop 1 indoor. I am experiencing some deviation to the left or right when going forward(positive pitch). When testing with FreeFlight Pro on Android I does not experience that problem. I tried using flat_trim, different speeds and durations. I would like to know if that's a library problem or maybe the android app does something extra we don't know. Thank you.

amymcgovern, 2018-09-04:
I see deviations when there are GPS issues (e.g. coverage going in and out).  Since you have a 1, you can turn the GPS off (e.g. indoor mode).  Have you done that?  With the 2, you have to cover the GPS unit but it makes it fly straight.

amymcgovern, 2018-09-05:
I haven't implemented the indoor mode since 1) I don't have a Bebop 1 and 2) It says it is deprecated.  But I assume you can do it on the app for the bebop 1.  If not, you can always do the approach we did which was to cover the front nose cone area (but not the camera) with tin foil.  It keeps the bebop 2 from locking onto GPS.

For the max_tilt, yes, do it before takeoff!

ManuelCostanzo, 2018-09-05:
Thanks for all. 

I could disable the gps using the andorid apk and everything works better.

The only thing that remains to review is the following: if I unplug and plug the battery again, I make a takeoff with this library, the drone does not remain stable. Then I use the android apk, I make a take off and it stays stable. I'll try this library again and now the takeoff is stable.

I think the application sends calibration commands to the drone that disappear when I unplug the battery.

Any ideas ?

amymcgovern, 2018-09-05:
I can implement the indoor command but I don't have a bebop one to test it on.  The code would be easy though.  And yes, rebooting (e.g taking out the batteries) likely is resetting things.  @Victor804 could you test the indoor command if I add it?

ManuelCostanzo, 2018-09-05:
If you want i can test it, i have a bebop 1 and i use it only indoor

Victor804, 2018-09-06:
Yes I could this weekend to confirm with @ManuelCostanzo that it works but I will be less able to participate in the project because I have returned to school.

amymcgovern, 2018-09-06:
It is uploaded into version 1.5.3.  And @Victor804 you have been a great contributor!  @ManuelCostanzo let me know if there are any issues.  It's hard to test since the command is ignored by the bebop 2.

ManuelCostanzo, 2018-09-06:
Hello and thanks ! I tried the indoor code but is stack on the while waiting for "sensor outdoor mode changed".

amymcgovern, 2018-09-06:
I can remove that wait.  I wasn't sure if it would send since it says the command is deprecated.

Honestly, try the tin foil hat thing.  We covered the front nose cone with tin foil before booting.  Solved all indoor GPS related issues with the bebop 2....

haydenhughes, 2018-09-08:
Just did some testing and the set_indoor is working flawlessly for me. Just make sure that you are in a well lit area to assist with the stabilization.

closed 2019-01-09 by amymcgovern