amymcgovern/pyparrot #105: Issue with demoMamboVisionGUI.py

PascPeli, 2018-09-27:
As the title suggests I have problems with the demoMamboVisionGUI.py on windows 10 with python 3.6.6. and vlc 3.0.3. I get a video stream very rarely (not perfect fluidity but better than nothing). All the other times the droneVisionGUI pops up but it has a black screen, no stream at all. Is there a workaround?

amymcgovern, 2018-09-27:
The mambo turns off the stream after one minute of not flying.  The solution is to either takeoff (turns the stream on) or reboot the mambo.  Either will restart the camera.

PascPeli, 2018-09-28:
> 
> 
> The mambo turns off the stream after one minute of not flying. The solution is to either takeoff (turns the stream on) or reboot the mambo. Either will restart the camera.

Thank you for your reply. I have taken this into account. Even when it is flying the video stream doesn't work.

amymcgovern, 2018-09-28:
That sounds like a hardware issue.  Also, I did have a windows 10 bug that was fixed after 1.4.30 so I assume you have the latest version.  But the hardware itself is running ffmpeg to decode the images

closed 2019-01-09 by amymcgovern